pelya/android-keyboard-gadget #21: keyboard partially not working

Brandhor, 2015-01-23:
I'm on n5 with cataclysm lollipop rom, elementalx kernel and touchpal keyboard
the mouse works fine but the keyboard is not fully working, for example enter and space works but the letters and numbers don't, also if I switch to the google keyboard my touches goes through it and I can see the touchpad area flashing like there was no keyboard at all

pelya, 2015-01-23:
It should work with Hacker's Keyboard:
https://play.google.com/store/apps/details?id=org.pocketworkstation.pckeyboard

I'll test with Google keyboard.

pelya, 2015-01-23:
It works okay with Google keyboard on Nexus 7 2012 with AOSP. I don't know
how to reproduce that bug.

Brandhor, 2015-01-23:
hacker keyboard works perfectly thanks

Brandhor, 2015-06-26:
the keyboard is not working again, I'm using a nexus5 with android 5.1 and elementalx 4, I've compiled the kernel myself with the hid patch and while the mouse works the keyboard doesn't, even with hacker keyboard, I've also tried another nexus5 with android 5.0 and elementalx 2 which has the patch builtin and the keyboard is not working either so maybe it's an app issue
I've also tried hid-gadget-test but neither the mouse nor the keyboard work with it, is there anything I can do to debug it?

Brandhor, 2015-06-27:
my bad it was actually a driver issue with my windows installation

closed 2015-06-27 by Brandhor