pelya/android-keyboard-gadget #23: remote-client suggestions

lars18th, 2015-02-19:
Hi Pelya,

First: Thank you for the VNC support! It's a must have! I'm testing it and it's working as espected. No errors found.

Realted to version 1.10 of remote-client, I have these suggestions:

1) Please, add the option to enable/disable the camera. Specifically I suggest to starting with camera off and provide a method to enable/disable it. When the camera is off, you only need to send a black screen (with a very low fps, like 1 update per second).

2) How to invoque the remote overlay menu? When I connnect using a VNC client, I can see some menu windows, but I don't know how to call them from the VNC client.

3) I also suggest to add a config option to autostart with the VNC server enabled. I feel this can be a high priority function, because my target scenerio is a remote mobile connected to a remote server, and in this scenario I don't have the option to access to the physical device.

I hope you can improve the current version when you have sufficient time. I really appreciate your effort. Thank you very much!

pelya, 2015-02-19:
You can invoke menu with Scroll Lock key. When you enable VNC server, it
will be launched on the next app start, and you can launch the app at boot
using Tasker for example. I'll add config to set different capture
dimensions in the next version.

lars18th, 2015-02-19:
Hi Pelya,

Thank you for the comments! :-)
After a new test I have these troubles:

1) My VNC Client isn't sending <SCROLL LOCK> keystrokes. Please, don't use the infamous key combination of serveral buggy KVM switches! I suggest that the user can select the hot-key for launch the menu.

2) If I CLOSE the App with VNC enabled (return to launcher and close it in the recent Apps List) and the restart the App, then the VNC server is OFF. Please, add an option to auto-start VNC server when opening the remote-client App.

Related to Tasker, I'm already using it. However I suggest to add the option to auto-launch at boot. It will be more easy for serveral users (the idea is a phone all the time connected to a remote server).

Finally, please, when in the next version you implement the option of multiple screen (video) resolutions, don't forget to support black screens without video! I really appreciate to enable/disable the camera capture.

Best!

lars18th, 2015-03-16:
Hi Pelya,

You will fix soon the bug of auto VNC start?

Regards!

lars18th, 2015-03-23:
Hi Pelya,

I hope soon you can fix current bugs of the VNC server.

Regards!

pelya, 2015-03-23:
Currently I'm short of free time, so I will probably get back to this app
in two-three weeks, sorry for the delay.
How exactly VNC does not sending keystrokes? It's working for me, I've
tried with KRDC and Android VNC viewer.
Mouse control still does not work so well near the edges, I'll fix that.

lars18th, 2015-03-24:
Hi Pelya,

Thank you for your response! :)
When you have time, please, consider these suggestions:

1) When I connect using the VNC client, isn't not possible to invoke your "hot key" (scroll lock) for show the menus. Please, add the option to configure this key. At time, I can only enable the menus using the touch screen in the mobile, and not from the remote client.

2) The autostart option of the VNC server don't works. Please repeat this: configure this option to enable, then reboot your mobile... check that when opening your app the VNC server don't starts.

3) Please, add an option to "disable the camera". It's so easy as it's only to send a simple black screen (time to time). I like to use the app for remote control, but I don't need the camera. This is slowering the VNC connection, but only for sending data that I don't need. The idea is a function for dinamically enable/disable the camera.

Finally, I have a suggestion for a new functionality. Please, consider it:

- Add an alternative video input. In the current app, you're reading the image from the camera. However, if you can add the option for reading from another source (http, udp or stdin) it will be very easy to use external inputs. My objective is use ffmpeg or vlc running in another device for sending the video of the screen (with a very low complexity). I feel this option can be easy to implement if you use raw format.

What you think about this?

Best!

pelya, 2015-03-24:
Autostart does work for me. Clean Nexus 7 2012, I've installed USB Keyboard
app from Google Play, flashed boot image inside the app, rebooted. Launched
the app, enabled VNC server, closed the app with Back key. Launched again,
VNC is started with the app. Rebooted Nexus 7, launched the app, VNC server
started automatically.
Maybe there's something wrong in your setup? Or maybe in mine?

I'll add points 1 and 3 in the next release.

As for external video source, I need some valid real-world use case, for
example forwarding video from IP camera, which means I have to buy IP
camera first. So, well, maybe, but don't expect this feature any time soon.

lars18th, 2015-03-25:
Hi Pelya,

Thank you for consider points 1 and 3 for the next release! :)
Also, can you please, add a menu option for "Exit" or "Close" the App?

Related to autostart: I uninstalled my previous version, redownload from the Play, and now it's works! Great!

Related to external video source, my user case is receive the screen (video) from a HDMI capture device. However, you idea of a IP camera is also very interesting. Moreover, the mechanism to receive the video can be the same. If you like to "test" some IP camera without purchase any, you can try some "Camera IP" software for Android. For exemple, this App is very useful: https://play.google.com/store/apps/details?id=com.pas.webcam
For receive the stream, you can use HTTP requests, or CURL. Several formats are supported (mp4, MJPEG, PNG, BMP... This can be a very good improvement!