pelya/android-keyboard-gadget #25: BeagleBone Black, successful compilation but unable to boot (Lacking MLO, u-boot.img, uImage, uEnv.txt, rootfs.tar.bz2)

Sumofwhich, 2015-03-26:
Warmest greetings everyone,

I've managed to compile the kernel (zImage) and the image (boot.img) following the steps. 
I then used win32diskimager to write the boot.img to the sdcard, but the beaglebone black couldn't boot up. 

I tried using rowboat's mkmmc-android.sh script to write the files to the sdcard, but I still need some other files such as MLO, u-boot.img, uImage, uEnv.txt, and rootfs.tar.bz2
<MLO> <u-boot.bin> <uImage> <boot.scr> <rootfs tar.bz2 >
https://gitorious.org/rowboat/external-ti_android_utilities/source/8d6bf9f66eeab6bba7f9828adf1e2ab194ee82d7:am335x/mk-mmc/README.txt#L35

I've tried googling for ways to generate these additional files, but most of the online resources only relate to kernels from rowboat (e.g. this tutorial https://github.com/csimmonds/bbb-android-device-files/tree/kk4.4-sdcard). 

Are there any possible methods for BeagleBone Black?

Sumofwhich, 2015-03-26:
I also did checkout rowboat-am335x-kernel-3.2, I tried manually editing the files (based on kernel-3.1.patch found here), but some of the lines are missing.

And I'm not sure if the build would be successful, please advise. Thanks!

pelya, 2015-03-26:
Hi.

Did you try to generate a build without this patch, using clean Beaglebone
repository? Did it boot?

Please attach here your build logs (you can use smth like
http://gist.github.com or http://pastebin.com

Sumofwhich, 2015-03-26:
Warmest greetings Pelya,

Thanks for your prompt reply :)
I will try to build it based on this tutorial (https://github.com/csimmonds/bbb-android-device-files/tree/kk4.4-sdcard) BUT using a clean repo and -b android-4.4.2_r1

Will let you know when I'm done and will attach the build logs. Thanks!

Sumofwhich, 2015-03-30:
Completed, everything is successful but couldn't boot up with the sdcard method, skipped SGX and Android VNC server

Logs:
https://gist.github.com/d879be890aa3220f0f41.git

pelya, 2015-03-30:
I didn't see anything suspicious in the log, so I cannot tell where is the
problem.

Sumofwhich, 2015-04-02:
When I turned on the power, all 4 leds are not lit up. I could only see red lines on the display (I'm using HDMI to DVI converter).

When I tried using aosp version 4.4.4_r1 with sgx drivers and android vnc server installed, all 4 leds are not lit up too. I could see red lines on display at first, but the android logo shows up and I manage to boot into the home screen.

Sumofwhich, 2015-04-02:
Ahh it's okay Pelya, I've managed to manually apply the changes in kernel 3.1.patch for the android aosp 4.4.4_r1 build, and then rebuild it. 

Now I can send keypresses by following your Scripting section! Thanks a lot!

conrad-heimbold, 2017-12-21:
Problem seems to have been fixed, therefore closing.

closed 2017-12-21 by conrad-heimbold