pelya/android-keyboard-gadget #27: BOOT Mode support

lars18th, 2015-05-25:
Hi Pelya,

Please, can you add BOOT Mode support in your patches for the kernel?

See this post for the patch: http://www.spinics.net/lists/linux-usb/msg125007.html

I hope you agree. ;-)
Regards.

pelya, 2015-05-25:
I thought USB keyboard already works in BIOS, and the patch already has
Boot interface subclass in USB descriptor. Does this patch fix an issue
with some particular device that you have?
Anyway, I'll be able to only patch Nexus 7 2012 kernel, because I don't
have other devices.

lars18th, 2015-05-25:
Hi Pelya,

Your patches describe the correct subclass (BOOT mode), but they DON'T work with bioses because the reasons described in the thread:

1) The USB specification describes that **ALL** HID devices start in REPORT mode (not in BIOS mode), then after the initial handshake is when it can requested to can change to BOOT mode. And yes, the report and boot modes can be equal (as in your code), but when a BIOS requests to change to BOOT mode, and the device respons with an error (like now), then it refuses to work with the device and remove it from the HID input list.

2) The current problem is in the driver implementation of the Linux(/Android) kernel, mainly with the lack of the HID "Set_Protocol" and "Get_Protocol" functions. Without this patch it's IMPOSSIBLE to change from the report protocol to the boot protocol, and this is a requiment to work with bioses.

Please, belive me... I test your software and patches with several bioses, KVMs, etc. I don't work at all for this specific problem: the misimplementation of these funtions. This patch partially solves the problem, so I suggest to incorporate in yours. If you can only test on Nexus 7 2012, don't care about it. I'll work as it's a small change with all devices.

Please, try it, and comment your results.
Regards.

pelya, 2015-05-25:
I've merged that patch to all kernel patches in the Git repo. I've built
Nexus 7 2012 boot image for Android 5.1.1, but for other kernel patches,
I've only checked that they can be applied and respective kernels will
compile. I've tested it and it does not break any functionality, please
reply if it improves anything on your setup.

I still remember about options for camera feed size and for redefining VNC
hotkey, I don't have much free time this month, but I'll implement them
eventually.

lars18th, 2015-05-26:
Hi Pelya,

Thank you for fast reply, and for the new patches! I'll try and post results.

Regards.

lars18th, 2015-05-27:
Hi Pelya,

A new version of the patch is available: http://permalink.gmane.org/gmane.linux.usb.general/126204

Sorry for not comment until this definitive version. This new version fixes the default mode and improves conformance with specifications. I hope this will be the last modification. And I belive it will be commited to the main kernel branch when sufficient testing is done.

Please, can you upgrade your patch as are only a minimal modification?
Thank you!

pelya, 2015-05-27:
Maybe next week