pelya/android-keyboard-gadget #29: hid-gadget-test qwerty azerty

letroll, 2015-06-24:
Hello, I have a bug with hid-gadget-test utility, when I run : echo a | /data/local/tmp/hid-gadget-test /dev/hidg0 keyboard, it's print "q" and vice versa 

I think to a qwerty azerty conversion... Can you help me with that?

pelya, 2015-06-24:
It's definitely because of AZERTY layout.

You can use some commandline conversion of characters, like

echo a | /data/local/tmp/busybox tr qwaz azqw |
/data/local/tmp/hid-gadget-test /dev/hidg0 keyboard

You will also need to copy busybox to /data/local/tmp

letroll, 2015-06-25:
Is it possible that I compile HID-gadget-test with the azerty layout? sources are available?

pelya, 2015-06-25:
Yes, here they are:
https://github.com/pelya/android-keyboard-gadget/tree/master/hid-gadget-test/jni
Run ndk-build from hid-gadget-test directory to recompile it.

letroll, 2015-06-25:
Thanks a lot I'll try that

pelya, 2015-09-11:
.val is scancode returned by keyboard hardware. There is a list of USB keyboard scancodes here:
http://hg.libsdl.org/SDL/file/tip/include/SDL_scancode.h
When you have non-QWERTY layout, you just have different key labels, but the key themselves usually return same scancodes - upper-left key has scancode 20, where would be 'Q' key on QWERTY keyboard, the next key has scancode 26 etc.

polarsbear, 2017-03-06:
I live in Belgium, So I use AZERTY, Thanks for the amazing utulity btw. 
If I redo the file for my layout, will you add it to the repo? So that others can use this and it is easy to find.