pelya/android-keyboard-gadget #30: Cyanogenmod 12.1 stuck - selinux

0uro, 2015-06-29:
Program starts and stays at "opening /dev/hidg0 and /dev/hidg1".

Found a closed issue from prodoomman, he told me to either write a selinux policy for hidg* or deactivate selinux. I can't write policys but deactivating it did the trick.

As the program already asks for root permissions this is quite annoying

pelya, 2015-07-01:
Good point.
I'll make it deactivate SELinux automatically in the next version.

R03LTjuh, 2015-07-24:
I still dont like the idea also in redhat selinux preinstalled when there are more to chose from.
So thats a yes please.

I like apparmor couse it keeps freedom of choice instead of bricking your distro by removal.