pelya/android-keyboard-gadget #33: Alternative camera input

lars18th, 2015-07-03:
Hi Pelya,

As you commented on:
http://github.com/pelya/android-keyboard-gadget/issues/20#issuecomment-73338202
the current implementation of the USB Keyboard app can compile on plain desktop Linux (with SDL).

So I suggest to implement a simple "alternative" input for the camera in this environment. The code of the camera class seems to be very compact. Then, why not include an FFMPEG rawvideo pipe for capture frames? Please, see some simple examples at:
http://dranger.com/ffmpeg/ffmpegtutorial_all.html

As you remember, my idea is to use the Android device as simple "USB gadget proxy" for send commands to "/dev/hidgX" ports. At time, I can do that with network pipes, and use your USB Keyboard app in a Linux machine. So now, I like to connect this app (in the workstation) to an IP Camera (using FFMPEG).

What you think about this "crazy" idea?
Thank you for read this!

pelya, 2015-07-03:
This can be done, but as always requires time, which I currently don't
have. Maybe in two-three weeks I'll get back to this app.

lars18th, 2015-07-03:
@pelya 
Thank you for the response. I appreciate your effort. ;-)