pelya/android-keyboard-gadget #35: Keyboard/mouse event

miniME89, 2015-08-08:
Hey,

i am currently using your patch for developing an android app where I had a problem with your definition of the keyboard event structure:

You are saying that the keyboard event is an array of 8 bytes. So I though the following command should also work for writing the letter "a" to the device:

```
echo -n -e "\x00\x04\x00\x00\x00\x00\x00\x00" > /dev/hidg0
```

This didn't work for me, so after some hours (!) of searching for any problems in my Java code, which executes this command, I used your (precompiled) example program *hid-gadget-test* to verify the actual output by directing the output into some file:

```
echo c | ./hid-gadget-test /mnt/sdcard/test.txt keyboard
```

which actually writes the following byte sequence into the file (hexdump):

```
0000 0400 0000 0000 0000 0000 0000 0000
```

So these are 16 bytes instead of the expected 8. If I change my echo command to output 16 bytes instead of 8 it works perfectly.

Tested on Nexus 7 2012 with self compiled kernel. I didn't use the mouse events, but I would assume the same problem there.

Drolmen, 2017-11-17:
Have you try control your pc by send mouse event ?

Mis012, 2018-11-10:
wtf?
0000 0400 0000 0000 <-- key down

0000 0000 0000 0000 <-- key up

2*8 = 16