pelya/android-keyboard-gadget #36: Configuring hid-gadget-test for different keyboard layouts

zeratoss, 2015-09-10:
Hey man great job!
I would love to use this with my Qwertz computer.
I saw https://github.com/pelya/android-keyboard-gadget/issues/29 and know that you have to recompile hid-gadget-test but how would you do that for different keyboard layouts?

pelya, 2015-09-11:
Will you be using hid-gadget-test utility or the GUI app? You can
reconfigure all keys inside GUI app.

zeratoss, 2015-09-17:
The hid gadget utility with the duvky script bash interpreter from
http://zx.rs/6/DroidDucky---Can-an-Android-quack-like-a-duck/

pelya, 2015-09-18:
Thanks for the link, I've added it to readme.

pelya, 2015-09-18:
Well you just put QWERTY keyboard near your QWERTZ, and note all keys that
are swapped from QWERTY layout. Then you swap the appropriate keys inside
hid-gadget-test sources, and recompile it.

zeratoss, 2015-09-23:
The problem is that some keys like "\" have no direct equivalent. On Qwertz it is "Alt Gr" + the key right of 0, but on Qwerty that combination does nothing, so i can't switch them.

tejado, 2016-05-24:
Check https://github.com/tejado/Authorizer/blob/master/app/src/main/java/net/tjado/authorizer/UsbHidKbd_de_DE.java for a german scancode mapping.

zeratoss, 2016-06-01:
@tejado thank you that looks really useful.
Can i just add missing keys like
`{.opt = "ß"				.val =0x2d},` ?

Is it different for values longer than four like 
`kbdVal.put("€",					new byte[] {0x40, 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00} );`?

tejado, 2016-06-01:
With the ß you can. Maybe for € there is no adjustment necessary as it should be just "echo right-alt e | ./hid-gadget-test /dev/hidg0 keyboard" or? (not tested)

VirtualThin, 2016-11-18:
@zeratoss  @tejado 

I have the same problem as zeratoss has. I tried to add the \ (german keyboard) but i didn't get it working.
Did you manage it?
Can i change: {.opt = "backslash",	.val = 0x31},? But which value?
@tejado  I've found this line in your repository for german layout: {0x02, 0x00, 0x1f, 0x00, 0x00, 0x00, 0x00, 0x00}  But i think, this is a too long. 

Any ideas?

tejado, 2016-11-18:
@VirtualThin 
it isn't too long, it is the complete 8 byte scancode like explained here: https://github.com/pelya/android-keyboard-gadget#how-it-works
So my table is a complete mapping table as the mapping like in https://github.com/pelya/android-keyboard-gadget/blob/a41e6e2511cd5d76bc7aa2c9aa6ea45e98f0b78c/hid-gadget-test/jni/hid-gadget-test.c is done dynamic but is difficult to change to support other languages.