pelya/android-keyboard-gadget #37: Kernel

PentesterGuy, 2015-09-18:
Hi, 

Thanks for sharing this great gadget :) 

But i want help about samsung galaxy trend plus s7580 ? 

How i should use this app ? 
I have rooted my phone but don't know about kernel ? 
I installed the usb keyboard from play store but it says i have to compile the custom kernel ? 
Pleae help ? 

Best Regards

PentesterGuy, 2015-09-18:
Does this kernel work ? 
https://github.com/SandPox/android_kernel_kylepro

thanks

pelya, 2015-09-18:
Most probably this kernel will not work. Your phone does not even have
subforum on XDA. I'm only providing kernels for Nexus 7 2012, because
that's the device I have, and I rely on people from XDA forums to build
kernels with my patch for other devices.

PentesterGuy, 2015-09-18:
Sir now what should i do ? 
help please

PentesterGuy, 2015-09-18:
Sir how this website use ? 
http://zx.rs/6/DroidDucky---Can-an-Android-quack-like-a-duck/

pelya, 2015-09-18:
They probably just used Nexus phone, patched kernels are available for both
Nexus 5 and Nexus 4.

PentesterGuy, 2015-09-18:
Sir how can i use patch for galaxy trend plus one ? 
I have downloaded the kernel from this link https://github.com/SandPox/android_kernel_kylepro
Now what should I do after this ?

pelya, 2015-09-18:
You apply the patch using command 'patch -p1 < patchfile.patch'. It's
available on every Linux distribution and on Cygwin.

PentesterGuy, 2015-09-18:
Then i am done ?

pelya, 2015-09-18:
Then you compile it, and assemble a boot image as described in the video,
and flash it to your phone.

PentesterGuy, 2015-09-18:
Ok thanks lemme try if I could i would really need your help thanks :)

PentesterGuy, 2015-09-18:
Sir please can you explain this two part below ? 
http://imageupl.net/?di=3QEP
http://imageupl.net/?di=R23K
Do i need to do this for my device ? 

thanks

PentesterGuy, 2015-09-19:
Sir while patching it throws this error : 
root@pentest:~/Android-Kernel/kernel-source# patch -p1 < /root/kernel-3.4.patch 
patch: **** Only garbage was found in the patch input.

thanks

PentesterGuy, 2015-09-19:
Sir i get error while patching :( 

root@pentest:~/Android-Kernel/kernel-source# patch -p1 < /root/kernel-3.4.patch 
patching file drivers/usb/gadget/Makefile
Hunk #1 FAILED at 55.
1 out of 1 hunk FAILED -- saving rejects to file drivers/usb/gadget/Makefile.rej
patching file drivers/usb/gadget/android.c
Hunk #1 succeeded at 53 with fuzz 2 (offset -21 lines).
Hunk #2 succeeded at 1073 with fuzz 2 (offset -1018 lines).
Hunk #3 FAILED at 2156.
Hunk #4 FAILED at 2481.
2 out of 4 hunks FAILED -- saving rejects to file drivers/usb/gadget/android.c.rej
patching file drivers/usb/gadget/f_hid.c
Hunk #7 succeeded at 403 (offset -9 lines).
Hunk #8 succeeded at 422 (offset -9 lines).
patching file drivers/usb/gadget/f_hid.h
patching file drivers/usb/gadget/f_hid_android_keyboard.c
patching file drivers/usb/gadget/f_hid_android_mouse.c

thanks

TranscranialRapist, 2015-09-22:
Using Note 4 N910T, is it possible to get this running for me?  I'm using Nadia custom kernel for 4.4 stock based rom DynamicKat. Linux is permissive.

pelya, 2015-09-22:
There was a patched kernel for your Note 4 model here:

http://forum.xda-developers.com/note-4-tmobile/orig-development/30-november-2014-n4-ael-kernel-t2961240

The download link is broken, it looks like the downloads have been moved
here: http://www.echoerom.com/ael-kernels/

PentesterGuy, 2015-09-22:
@pelya  Why not answering my questions ? 

:(

pelya, 2015-09-22:
Try to apply another patch for different kernel version from the website,
until there will be no failed chunks. If that won't help, apply failed
chunks by hand, that requires some knowledge of C.

PentesterGuy, 2015-09-22:
Sir the kernel of my device is 3.4.5 and the kernel patch is 3.4 does it matters?

PentesterGuy, 2015-09-23:
Sir please can you fix the kernel-3.4.patch for kernel version 3.4.5 ? please

thanks

PentesterGuy, 2015-09-24:
@pelya bro please help I have tried many times I patched manually the kernel but again problem with compilation please help :( 

thanks

PentesterGuy, 2015-09-25:
Here is the Image http://imageupl.net/?di=FM06 Help!!!

conrad-heimbold, 2017-11-07:
> Sir the kernel of my device is 3.4.5 and the kernel patch is 3.4 does it matters?

Depends on how much C code changed. Try to fix the patch errors manually by opening every $FILENAME and $FILENAME.rej in parallel. Compile the kernel again afterwards. 

Your original question has been answered. 

If you ask again and again, you will only make people angry.

closed 2017-11-07 by conrad-heimbold