pelya/android-keyboard-gadget #38: Hid utility quiet switch?

Gemok1980, 2015-10-18:
Hi 
I've managed through a fair amount of trial and error to write a simple script that reads a series of key presses from a txt file ( basically Ctrl c, Alt tab, ctrl v, next entry)and repeats a few hundred times. 
I put a echo counter in the code but can never see its output as the phone screen is always filled with the possible key entries. 
Is there a quiet switch to stop displaying the keys and only show my echo commands? 
Thanks 
For the very useful utility
Chris

pelya, 2015-10-18:
I've fixed that, please do git pull.

Gemok1980, 2015-10-19:
Cool thanks I'll do that. 
Again thanks for the utility its saving me loads of time at work. I set it up last thing and when I come in in the morning its done a couple of thousand for me :)

closed 2015-10-19 by Gemok1980