pelya/android-keyboard-gadget #39: Not working in parallel with rndis

mame82, 2015-10-21:
I need to bring a configuration like the below one this to work:

echo adb,mtp,hid,rndis > /sys/devices/virtual/android_usb/android0/functions

but I could only end up with either "adb,mtp,hid" enabled or "rndis,hid" enabled.

The second case would be enough, but hid isn't working although shown.
Does it depend on "mtp" ? If yes - do you know a configuration to run "mtp" and "rndis" at the same
time and thus make hid work again?

In fact I need a rndis adapter running, which responds to client-actions triggered by hid commands,
but I'm only able to run one of both usb.

pelya, 2015-10-21:
Hi. I don't have experience with rndis, so I cannot help, sorry. My Nexus 7
cannot share internet connection over USB port, so I cannot even reproduce
that.

HID driver does not depend on MTP.

mame82, 2015-10-22:
In the end it is working, the issue came down to a mistake made by myself.

I used commands like "echo rndis,mtp,adb,hid > /sys/devices/vitual.../functions" which couldn't work, because hid is loaded twice and thus crashes (documented in another issue here).
To make things work I have to echo out "rndis,mtp,adb" only. I don't know if it is important to place rndis in the beginning, because I have been reading that it is impotant to load rndis driver first on composite devices. A working idVendor:idProduct combination was 04e8:6863 (Samsung RNDIS).

so issue could be closed - thx for your great work.

conrad-heimbold, 2017-11-07:
> In the end it is working, the issue came down to a mistake made by myself.

Therefore closing.

closed 2017-11-07 by conrad-heimbold