pelya/android-keyboard-gadget #40: jfltexx, latest cm12.1, latest alucard kernel

dvd83i, 2015-10-29:
Hello. I performed all the steps in the xda forum installing the alucard kernel but when I plug the device to my PC, the USB KEYBOARD app crashes, but if I unplug, the app returns to work. Thank you

PentesterGuy, 2015-10-29:
See If you have debugging mode enabled :D

pelya, 2015-10-29:
Could you please collect ADB logs when it crashes?

dvd83i, 2015-10-29:
Hi! I just formatted the device and reinstalled cm12.1, alucard kernel, gapps and USB keyboard app from playstore; I enabled USB debugging and USB HID STweaks and restart the device; I connected it to PC and start the app. Same problem! When I touch the touchpad box or press any key on the display, it crashes!

dvd83i, 2015-10-29:
Finally it works fine! I istalled latest usb driver from samsung! TnX! 5 Stars!

closed 2015-10-29 by dvd83i