pelya/android-keyboard-gadget #43: oneM8 build

rudius, 2015-11-30:
thanks for the patch. I applied it on the dirtyunicorn kernel 3.4.109 for the HTC oneM8.  -> https://github.com/DirtyUnicorns/android_kernel_htc_msm8974.

Every thing looked good excepts that /dev/hidg* are not created.

i have the following if that can help:
root@htc_m8:/ # cat /proc/devices | grep hidg                                  
229 hidg

root@htc_m8:/ # cat /sys/class/android_usb/android0/functions                  
mtp,adb

I tried to create with mknod but i had "no such device or address"

Is there something that i could do to make it work?

pelya, 2015-11-30:
Could you please try

echo 0 > /sys/class/android_usb/android0/enable
echo mtp,adb,hid > /sys/class/android_usb/android0/functions
echo 1 > /sys/class/android_usb/android0/enable

hid should be added to that list automatically inside driver, but for some
reason this does not happen on your kernel.

thanks for the patch. I applied it on the dirtyunicorn kernel 3.4.109 for
the HTC oneM8. ->
https://github.com/DirtyUnicorns/android_kernel_htc_msm8974.

Every thing looked good excepts that /dev/hidg* are not created.

i have the following if that can help:
root@htc_m8:/ # cat /proc/devices | grep hidg

229 hidg

root@htc_m8:/ # cat /sys/class/android_usb/android0/functions

mtp,adb

I tried to create with mknod but i had "no such device or address"

Is there something that i could do to make it work?

rudius, 2015-11-30:
i don t have enabled

root@htc_m8:/ # echo 0 > /sys/class/android_usb/android0/enable                
tmp-mksh: can't create /sys/class/android_usb/android0/enable: No such file or directory

rudius, 2015-12-01:
i dont have CONFIG_USB_G_HID enabled in my kernel. Could it be some configuration problems?

pelya, 2015-12-01:
Yes, probably.

rudius, 2015-12-01:
i have currently android composite gadget selected in my conf. It seems i can t have both (HID gadget and android composite gadget) I assume the correct one is android composite gadget?

pelya, 2015-12-01:
Yes, you need Andrid composite gadget, the driver is compiled with it.

rudius, 2015-12-02:
HTC has modified the code to manage differently the gadget driver. I have made it work on my phone. The patch should be good with all kernel for it (HTC oneM8). I will close the issue. Let me know if you want to add the patch to your repo.
And a big thanks for your work! It s awesome.

closed 2015-12-02 by rudius

pelya, 2015-12-02:
If you publish your compiled kernel anywhere, like XDA, I would like to
mention it in the readme in my repo.

tejado, 2016-03-02:
Is this patch now somewhere available? If would be great if you can provide it. 
And would it work also for the htc one m7?

Thanks!

tejado, 2016-10-21:
push...
@rudius It would be really great if you can provide the patch! Thanks!

tejado, 2016-10-24:
FYI:
I solved it now by myself. PR: https://github.com/pelya/android-keyboard-gadget/pull/58

AngryBananer, 2017-04-11:
Could someone please upload the compiled kernel for the M8? I tryed to do it myself but it didn't worked....

tejado, 2017-04-12:
Yes, I can do this but I will need few days.

AngryBananer, 2017-05-05:
@tejado, when are you going to upload the kernel ?

NTuberz, 2023-08-28:
i know im really late to this, but has anyone compiled one? At least give me a link if you find anything pls and thank u