pelya/android-keyboard-gadget #44: USB Keyboard app - depth issue?

jcadduono, 2016-02-03:
I'm using Google Keyboard from the app store with your USB Keyboard app.
The touchscreen and everything is completely usable and working, but when I use the button to pop the keyboard up, I can't type on it.
All touch actions go straight through the keyboard and onto the touchpad or menu buttons.

Also, it'd be lovely if you could the possibility to pop the keyboard out in portrait mode if screen rotation enabled (or even a menu option) as it's a struggle for me to use a keyboard in landscape on a 6" screen. =P

pelya, 2016-02-03:
Please install Hacker's Keyboard from Google Play, ot should work better
with this app.

jcadduono, 2016-02-03:
Yeah that works fine. Weird that pressing back button goes to another keyboard entirely. I suppose I can just switch to hacker's keyboard app when I need it. Would be really nice to get portrait mode working though.