pelya/android-keyboard-gadget #45: kernel for note 4 sm-n910t

rathisarov, 2016-02-06:
Hi,
I'm using Galaxy note 4, SM-N910T.
Currently using beastmode kernel
http://forum.xda-developers.com/note-4-tmobile/development/kernel-beastmode-stock-1-0b-n910t3-t3238775/page45#post65175153
Can anyone help me compiling a patched kernel, or help me some guidance, if I can do it myself. 
Need some noob level instructions.
thank you.

pelya, 2016-02-10:
The only kernel for Note 4 with my patch I've found is this:
https://forums.kali.org/showthread.php?25009-Working-Samsung-Note-4-%28N910F%29-Support-for-Lollipop
You can try to compile your own kernel:
https://github.com/pelya/android-keyboard-gadget#compiling-kernel

conrad-heimbold, 2017-11-07:
Question has been answered, no answer this 1 year; therefore closing.

closed 2017-11-07 by conrad-heimbold