pelya/android-keyboard-gadget #46: Guidance for an aarch64 sony kitikami platform kernel source ?

moonbutt74, 2016-02-22:
pelya,

hi, the issue during compile is this;

```
drivers/built-in.o: In function `hid_function_cleanup':
/storage/sxZ5_dev/kernel/wafflez/drivers/usb/gadget/android.c:2979: undefined reference to `ghid_cleanup'
/storage/sxZ5_dev/kernel/wafflez/drivers/usb/gadget/android.c:2979:(.text+0x369dfc): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `ghid_cleanup'
drivers/built-in.o: In function `hid_function_init':
/storage/sxZ5_dev/kernel/wafflez/drivers/usb/gadget/android.c:2974: undefined reference to `ghid_setup'
/storage/sxZ5_dev/kernel/wafflez/drivers/usb/gadget/android.c:2974:(.text+0x369e18): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `ghid_setup'
drivers/built-in.o: In function `hid_function_bind_config':
/storage/sxZ5_dev/kernel/wafflez/drivers/usb/gadget/android.c:2986: undefined reference to `hidg_bind_config'
/storage/sxZ5_dev/kernel/wafflez/drivers/usb/gadget/android.c:2986:(.text+0x372d9c): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `hidg_bind_config'
/storage/sxZ5_dev/kernel/wafflez/drivers/usb/gadget/android.c:2992: undefined reference to `hidg_bind_config'
/storage/sxZ5_dev/kernel/wafflez/drivers/usb/gadget/android.c:2992:(.text+0x372dc0): relocation truncated to fit: R_AARCH64_CALL26 against undefined symbol `hidg_bind_config'
Makefile:801: recipe for target 'vmlinux' failed
```

I am working with sony kitikami platform kernel source 
VERSION = 3
PATCHLEVEL = 10
SUBLEVEL = 49
EXTRAVERSION =
NAME = TOSSUG Baby Fish

toolchain: aarch64-linux-android-4.9-kernel/bin/aarch64-linux-android-
I started from your kernel-3.10-nexus9.patch, as it seemed the closest starting point.
I am a little lost and still new to aarch64. Help !! =D

Thanks.
m

moonbutt74, 2016-02-23:
pelya,

hi again, UPDATE, the patch kernel-3.10-nexus6.patch
appears to do for the Sony Xperia Z5 kernel source at the moment,
I am running another build from zero just to see if i have it right.

thanks
m

pelya, 2016-02-23:
I never tried my patch with 64-bit kernel, so you are on your own here.

Does this kernel work?

moonbutt74, 2016-02-23:
Yeah it seems to but getting the patching procedure correct is proving a little diffcult. =/
My second build porked it with mssing files , so i git reset --hard and am going again.

moonbutt74, 2016-02-23:
pelya,

okay, build 4 and i can confirm kernel-3.10-nexus6.patch working with the aforementioned kernel source and toolchain, although i haven't tried any of the kali toys yet, the basic functionality of your app is go, mouse,keyboard. It doesn't appear to be aarch64 related at all, I just used the wrong patch at the beginning. This device uses the /dev/block/bootdevice/by-name/  setup which is REALLY annoying. =p. One final question, I am LOUSY with creating patches correctly, do i need to create one or will you just add a note to the existing one?

Thanks again.

m

pelya, 2016-02-23:
Ideally I would like a compiled boot.img compatible with some version of
Android (Kali is nice and all, but still cannot do voice calls).

Patch is not so important for me, I already got five of them in the repo,
doing the same thing, and I can only test one.

moonbutt74, 2016-02-23:
pelya,

Okay cool,
Sony_XperiaZ5p_HID_E6853_boot.img
https://www.androidfilehost.com/?fid=24421527759886007

Android version 5.1.1 SELinux permissive
Linux kali 3.10.49-bandersnatch_sxZ5_P_HID #1 SMP PREEMPT Tue Feb 23 02:54:49 EST 2016 aarch64 GNU/Linux

m

closed 2016-02-23 by moonbutt74

pelya, 2016-02-23:
I've put it into the repository, thanks.

joeisgood99, 2016-11-18:
Hi ghost/Moonbutt47. 
I am attempting to port this over to the stock Sony Kitakami Z5C kernel, as part of a wider NetHunter port. As you have done this for the Z5P I would be interested in looking at your work. I keep getting a persistent error while compiling. As you have changed your github account since this issue was closed Im leaving this comment here to see if I can locate you.
build env : Ubuntu 16.04
T/C GCC aarch64-linux-android-4.9
error: Makefile:798: *** multiple target patterns.  Stop.
which refers to script:   scripts/link-vmlinux.sh refereed to as $(vmlinux-deps)... I would love to look at your source code if it has been uploaded.
Regards

Joeisgood99.