pelya/android-keyboard-gadget #47: Freezes when running command

nnvt, 2016-02-29:
Hey!

Just patched and built a kernel for my zenwatch 2 with this patch and now the files /dev/hidg0 and /dev/hidg1 exist (i have chmodded them) but when I'm trying to run a preview command from your script like "echo a | ./hid-gadget-test /dev/hidg0 keyboard" it just freezes and does nothing at all :/

Any idea on how to fix this?

Thanks!

pelya, 2016-03-02:
Maybe try to disable SELinux? Does your watch have USB port to plug into PC?

simondankelmann, 2018-01-25:
You could also try connecting your Phone as Midi-Device before running your command.
Or try: setprop sys.usb.config "charging" before. Somehow ADB seems to interfere with the HID functionality (at least on my device)