pelya/android-keyboard-gadget #48: Not understanding readme

Leedl3, 2016-04-28:
Hi, I am trying to port this patch to the Moto E 2nd Gen "surnia". The kernel piece is fine its the boot.img piece I am skeptical about. This will need to be installed using fastboot right? Also the command: repo init -u https://android.googlesource.com/platform/manifest -b android-4.4.2_r1 . What would I need to change the android-4.4.2_r1 to? I am running cm 13 witch is android version 6.0.1. I do apologize if the answer to these questions are simple but I want to be sure that I don't mess up especially since this is the boot.img file we are talking about.

Thank you
-Leedl3

pelya, 2016-04-28:
These instructions are for AOSP. If you are using cyanogenmod, first you
need to compile cyanogenmod itself:
https://wiki.cyanogenmod.org/w/Build_for_surnia
Then you patch the kernel, recompile your cyanogenmod, and take the
resulting boot.img, which you can flash using fastboot.
You don't need system.img and other stuff, but unfortunately it will
compile everything, so that requirement to have 60Gb free space still
applies.

And if you succeed, please give me compiled boot.img.

Leedl3, 2016-04-28:
Ok, thank you!

conrad-heimbold, 2017-11-07:
Newer tutorial: https://wiki.lineageos.org/devices/surnia/build

closed 2017-11-07 by conrad-heimbold