pelya/android-keyboard-gadget #49: compiling issues

tim241, 2016-04-29:
hi I try to patch my Samsung galaxy core 2 (SM-G355HN) 3.10 kernel and I get 3 errors while patching: http://pastebin.com/PfybXQU0
and while compiling: http://pastebin.com/jtiaeKUY
I hope you can help me
Regards,
Tim241

pelya, 2016-05-01:
You will need to fix those patch errors first.
Where can I download that kernel sources?

tim241, 2016-05-02:
You can find the Kernel Source here: http://goo.gl/hlhnvs
Regards,
Tim241

pelya, 2016-05-18:
Here's a patch for your kernel, I only tested that it compiles, I did not try to run it.

https://gist.github.com/pelya/548e981523e5b737e80270340af49ccf

tim241, 2016-05-18:
I will test it tomorrow(removed ubuntu installed windows 7 so I need to dualboot tomorrow) thanks for all the work :-)

tim241, 2016-05-19:
I builded the zImage but how can I compile the boot.img on my Samsung galaxy core 2?(tried flashing the zImage trough Odin but it got stuck on NAND write start several times)

pelya, 2016-05-19:
You need boot.img, you cannot flash zImage directly. You get boot.img when
you compile full Cyanogenmod for your phone, but it's huge.
You can also try to download boot image from your phone boot partition,
unpack it, and repack with your own boot.img, however this methid never
worked for me.
Ask people on XDA how are they doing that:
http://forum.xda-developers.com/t/galaxy-core-2

tim241, 2016-05-19:
ok I contacted someone that builded cyanogenmod 11 for this device now I am waiting for a reaction. And I tried to unpack stock boot.img and replace zImage no luck :-(. Thanks for creating that patch file .

tim241, 2016-05-20:
Icreated an boot.img by having stock boot.img and unpack it and replace zimage qith the zimage but now my phone is stuck at the bootkogo :+1: I have fortunally a init.d boot.img on my phone but I really wish I will get this to work

pelya, 2016-05-20:
> zimage qith the zimage but now my phone is stuck at the bootkogo 👍 I

Try first to replace it with unpatched kernel image, it should at least
boot. You can also boot a stock image, and check kernel logs from last boot
with a command

cat /proc/last_kmsg

tim241, 2016-06-07:
Sorry for late reaction: 
(I can flash zImage with anykernel1(modified it), finally!!))

(I still cant boot but I copied the last_kmsg from recovery to sdcard(after booting with the kernel that did not work))

here is the last_kmsg(boot log file):  https://goo.gl/q6QlGA

I hope you can still help me :+1: 

Regards,
Tim Wanders // Tim's Tech

pelya, 2016-06-07:
I don't have an idea what is happening here. The kernel seems to boot fine,
it loaded hid keyboard and hid mouse drivers, according to log, then it
works for 10 more seconds, enters low power mode and probably freezes,
there is no crash in the log. Then it jumps into recovery mode, I assume
that's when you reboot it with a button.

Can you boot with unmodified kernel, generated the same way, but without
the patch?

tim241, 2016-06-10:
yes I will do that tomorrow, I will download the repo on my network drive from 2 terabyte. and then try to compile the boot.img from your instructions if it is not working then, I will try to do it with stock one and build that one.

tim241, 2016-06-11:
I tried building(stock) the zImage and then flashing it. No luck. I noticed that the compiled zImage is much smaller than the one in the stock boot.img.

tim241, 2016-06-20:
someone patched and compiled the kernel for me: /dev/hidg0: No such file or directory how to fix this?

tim241, 2016-06-20:
sorry but a friend patched the kernel and compiled it but the hidg files does not exist in /dev any idea?

pelya, 2016-06-20:
This means that the patch was not applied correctly, or you did not flash
the boot image correctly. Files /dev/hidg0 and /dev/hidg1 should be present
whenever you are plugging your phone into PC.
Can you see other files in /dev ? Newer versions of Android prevent regular
non-root apps from getting contents of some directories, like /dev.

tim241, 2016-06-21:
I can see other files in /dev but I use terminal emulater with su so I dont think that is a problem hmm
I flashed it correctly(can see other kernel info in settings) and the friend said that there where no problems while patching hmm there are no hidg files when I plug it into the PC with windows 10 and ubuntu 64-bit hmm

ih24n69, 2016-06-21:
@pelya I done patch(manual patching) and build kernel with no error. So what wrong which cause hidg not present on /dev folder?
I was build for Core 2 device.

tim241, 2016-06-21:
I flashed your kernel ih24n69 proof in screenshot and here in adb proof too:
[image]
`C:\>adb shell
shell@kanas3gnfc:/ $ su
root@kanas3gnfc:/ # cd /dev
root@kanas3gnfc:/dev # ls
CHNPTYW0
CHNPTYW1
CHNPTYW10
CHNPTYW11
CHNPTYW12
CHNPTYW13
CHNPTYW14
CHNPTYW2
CHNPTYW3
CHNPTYW4
CHNPTYW5
CHNPTYW6
CHNPTYW7
CHNPTYW8
CHNPTYW9
__properties__
alarm
android_adb
ashmem
batterylifeextender
binder
block
board_type
bus
console
cpu_dma_latency
cpuctl
cpw
device-mapper
dispc
fscklogs
full
fuse
graphics
i2c-0
i2c-1
i2c-2
i2c-3
i2c-6
i2c-7
i2c-8
input
ion
kmem
kmsg
log
loop-control
mali0
mem
mtp_usb
network_latency
network_throughput
null
pipe
ppp
ptmx
pts
random
rfkill
rpipe
rtc0
slog_w
snd
socket
spipe_w0
spipe_w1
spipe_w2
spipe_w3
spipe_w4
spipe_w5
spipe_w6
spipe_w7
spipe_w8
sprd_dma_copy
sprd_gsp
sprd_isp
sprd_jpg
sprd_rotation
sprd_scale
sprd_sensor
sprd_vsp
stty_w0
stty_w1
stty_w10
stty_w11
stty_w12
stty_w13
stty_w14
stty_w15
stty_w16
stty_w17
stty_w18
stty_w19
stty_w2
stty_w20
stty_w21
stty_w22
stty_w23
stty_w24
stty_w25
stty_w26
stty_w27
stty_w28
stty_w29
stty_w3
stty_w30
stty_w31
stty_w4
stty_w5
stty_w6
stty_w7
stty_w8
stty_w9
tgt
tty
ttyGS0
ttyGS1
ttyGS2
ttyGS3
ttyS0
ttyS1
ttyS2
ttyS3
tun
uhid
uinput
umts_router
urandom
usb_accessory
vbpipe1
vbpipe2
vbpipe3
vbpipe6
video0
xt_qtaguid
zero
root@kanas3gnfc:/dev #`

pelya, 2016-06-21:
What are the contents of file /sys/class/android_usb/android0/functions ?
It should have 'hid' inside. Could you please execute following commands:

echo 0 > /sys/class/android_usb/android0/enable

echo mtp,adb,hid > /sys/class/android_usb/android0/functions

echo 1 > /sys/class/android_usb/android0/enable

tim241, 2016-06-21:
now if I try it it works but it says killed and does not send the keys hmm maybe I need to chmod it right?

tim241, 2016-06-21:
hmm if I try to chmod it still says /dev/hidg0 not found

pelya, 2016-06-21:
The app launches chmod itself, but it needs /dev/hidg0 to be present. Could
you try the hid-gadget-test command line tool, as described here:
https://github.com/pelya/android-keyboard-gadget/blob/master/README.md#scripting

tim241, 2016-06-21:
I am doing that but it just says: killed
I am doing this command: echo a | ./hid-gadget-test /dev/hidg0 keyboard

tim241, 2016-06-21:
and if I do this command: echo 1 > /sys/class/android_usb/android0/enable my device will be stuck I need to reboot to fix that

pelya, 2016-06-21:
Okay, this means that the patch does not work properly, and I don't have
Galaxy Core 2 to fix the issue on my side.

Could you please copy here the contents of /proc/last_kmsg after the device
is stuck and rebooted?

and if I do this command: echo 1 > /sys/class/android_usb/android0/enable
my device will be stuck I need to reboot to fix that

tim241, 2016-06-21:
yea hold on

tim241, 2016-06-21:
ow after that command my phone is REALLY slow it was not hanging it is just REALLY slow. I cant acces my phone now from my PC strange http://www.mediafire.com/download/qzg093y26qp2eiw/last_ksmg

tim241, 2016-06-21:
ow now I can acces it again hmm

tim241, 2016-06-22:
Hi guys thanks for your help I really found everything helpfull but I think you guys need to do other important things You guys are wasting too much time for me. I will say: Thank you guys for your help but just don't help me anymore I found an other device that can do the same :-/ so Thanks for your help guys but I don't need it anymore, I will close this issue and move on
Kindly Regards,
Tim Wanders

closed 2016-06-22 by tim241

pelya, 2016-06-22:
Okay...

tim241, 2016-07-04:
@pelya : 
@ih24n69 ported it and sended me a message it works but: https://github.com/anbud/DroidDucky
the duckyscript does not work it says:
`root@kanas3gnfc:/storage/extSdCard/DroidDucky-master # ls
README.md
droidducky.sh
hid-gadget-test
test.dd
root@kanas3gnfc:/storage/extSdCard/DroidDucky-master sh droidducky.sh test.dd                           <
Loading payload code.
droidducky.sh: line 411: 11688 Done                    echo "$last_cmd"
     11689 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11692 Done                    echo "$kbcode"
     11693 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11695 Done                    echo "$kbcode"
     11696 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11698 Done                    echo "$kbcode"
     11699 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11701 Done                    echo "$kbcode"
     11702 Killed                  | ./hid-gadget-test $kb > /dev/null
Opening command prompt.
.5) Parse error: Unexpected ENTER
: syntax error: invalid arithmetic operator (error to")n is "
'sleep: invalid number '100
Sending the message.
droidducky.sh: line 411: 11710 Done                    echo "$kbcode"
     11711 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11713 Done                    echo "$kbcode"
     11714 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11716 Done                    echo "$kbcode"
     11717 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11719 Done                    echo "$kbcode"
     11720 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11722 Done                    echo "$kbcode"
     11723 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11725 Done                    echo "$kbcode"
     11726 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11728 Done                    echo "$kbcode"
     11729 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11731 Done                    echo "$kbcode"
     11732 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11734 Done                    echo "$kbcode"
     11735 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11737 Done                    echo "$kbcode"
     11738 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11740 Done                    echo "$kbcode"
     11741 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11743 Done                    echo "$kbcode"
     11744 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11746 Done                    echo "$kbcode"
     11747 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11749 Done                    echo "$kbcode"
     11750 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11752 Done                    echo "$kbcode"
     11753 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11755 Done                    echo "$kbcode"
     11756 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11758 Done                    echo "$kbcode"
     11759 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11761 Done                    echo "$kbcode"
     11762 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11764 Done                    echo "$kbcode"
     11765 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11767 Done                    echo "$kbcode"
     11768 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11770 Done                    echo "$kbcode"
     11771 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11773 Done                    echo "$kbcode"
     11774 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11776 Done                    echo "$kbcode"
     11777 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11779 Done                    echo "$kbcode"
     11780 Killed                  | ./hid-gadget-test $kb > /dev/null
droidducky.sh: line 411: 11782 Done                    echo "$kbcode"
     11783 Killed                  | ./hid-gadget-test $kb > /dev/null
root@kanas3gnfc:/storage/extSdCard/DroidDucky-master #`
He does not respond to the project so I hope you can help me.

pelya, 2016-07-04:
Does the hid-gadget-test tool work? Does the app work?

I never tried this droidducky interpreter. Maybe you need to run it from
busybox shell instead of native Android shell, which is limited.

tim241, 2016-07-04:
Yes the app and the hid-gadget test work

tim241, 2016-07-04:
I did sh instead od bash I did bash here: error: here is the output with bash: http://pastebin.com/nvCMEppy

tim241, 2016-07-04:
I fixed it by dos2unix

pelya, 2016-07-04:
Haha okay

ih24n69, 2016-07-05:
@pelya thank you youre patch for Core 2 it work :) .

tim241, 2016-07-06:
@pelya @ih24n69 one question if I check the supersu logs(I created my own payload app that will work with hid)
it says:
`FIX` ME! Implement ttyname() `bionic/libc/bionic/stubs.c:360`
can you try to fix that above error?
Anyway the hid works good +1

pelya, 2016-07-06:
I will not fix warnings in your Android libc, and this warning is fixed in
Android Marshmallow, so... update your phone?

tim241, 2016-07-07:
Okejj. It is just a error so it is not so bad anyway HID is so cool.