pelya/android-keyboard-gadget #53: add keystrokes for app

tim241, 2016-07-09:
hello @pelya I have a question:
I am creating a hid app but I don't want to work with the hid-gadget-test file. Is it possible to build it into my app to send keystrokes from your hid patch?(I am using xamarin(mono for android/ C# for android)
Regards,
Tim

pelya, 2016-07-09:
Yes, you just open /dev/hidg0 as a regular file, and write into it. Here's
a piece of code that does this:
https://github.com/pelya/android-keyboard-gadget/blob/master/hid-gadget-test/jni/hid-gadget-test.c#L178
The format of the bytes sent is described in readme:
https://github.com/pelya/android-keyboard-gadget#how-it-works

Because of SELinux, Android app cannot open the file directly, so it
launches su and pipes data through it, something like FILE *pipe =
popen("su"); fprintf(pipe, "cat > /dev/hidg0\n"); then write keycodes to
pipe, and don't forget to fflush().

tim241, 2016-07-09:
If Selinux is diabled I
can do someting like this right?:
Echo 0x01 | /dev/hidg0

pelya, 2016-07-09:
Yes you can.

tim241, 2016-07-10:
can you give me some more details about that " FILE *pipe =
popen("su"); fprintf(pipe, "cat > /dev/hidg0\n"); then write keycodes to
pipe, and don't forget to fflush()."? the command does not seems to work. ANd I am not understanding it.

pelya, 2016-07-10:
It's a piece of C code, I don't know how are you doing file writing in C#.
Open /dev/hidg0 and write into it chunks 8 bytes each, flush the buffer
after each write, that's it.

tim241, 2016-07-10:
if I create a test file(located on sdcard with: 0x0a in it) if I do a root shell and then: cat /sdcard/test > /dev/hidg0 but then it spams an random key from my keyboard, any idea?

tim241, 2016-07-10:
or can you explain how to do it with the terminal? Because I can easilly send shell commands like cat in su.

pelya, 2016-07-10:
Yes, you can copypaste a text file.
Please read the keyboard protocol description at
https://github.com/pelya/android-keyboard-gadget#how-it-works
You send 8 bytes at once, not 1, and the first byte is special.

tim241, 2016-07-11:
I still don't understand it :-(

tim241, 2016-07-13:
nevermind, I will use the hid-gadget test utillity :-)

closed 2016-07-13 by tim241