pelya/android-keyboard-gadget #54: Android Lollypop : hid_function_bind_config: hid_function_bind_config keyboard failed: -19

ajit1433, 2016-08-05:
I am trying to add HID functionality to my device but i keep getting the **hid_function_bind_config** failed error.

Below is output while booting device...

[    3.782222] adb_android_function_enable ...1
[    3.786518] ==kcj enable_store dev->enabled:1
[    3.791609] ==kcj enable_store
[    3.794741] ==kcj enable_store dev->enabled:0
[    3.799429] ==kcj enable_store
[    3.802525] ==kcj enable_store..2
[    3.805896] adb_android_function_enable ...1
[    3.810220] ==kcj enable_store dev->enabled:1
[    3.817129] adb_open
[    3.819438] hid keyboard
[    3.822014] hidg_bind: creating device d6f3e700
**[    3.827827] android_usb gadget: hidg_bind FAILED**
**[    3.832534] hid_function_bind_config: hid_function_bind_config keyboard failed: -19**
**[    3.841698] android_bind_enabled_functions: hid failed**

Please help.

pelya, 2016-08-05:
What device is this? Who built the kernel?

ajit1433, 2016-08-06:
Hi,

Following are the details:

*Android OS Version: *
ro.build.version.sdk=17
ro.build.version.codename=REL
*ro.build.version.release=4.2*
......
ro.product.name=rk2928sdk
ro.product.device=rk2928sdk
ro.product.board=rk2928sdk
ro.product.cpu.abi=armeabi-v7a
ro.product.cpu.abi2=armeabi
....
ro.board.platform=rk2928
# ro.build.product is obsolete; use ro.product.device
ro.build.product=rk2928sdk

*Kernel Version:*Linux version 3.0.36+ (t@ubuntu) (gcc version 4.6.x-google
20120106 (prerelease) (GCC) ) #576 PREEMPT Sat Aug 6 12:43:37 IST 2016

Thanks and Regards

Ajit Pratap Singh / Graduate Student
9718843393/ Ajit1433@iiitd.ac.in <ajit1433@iiitd.ac.in>
<ajit1433@iiitd.ac.in>

<ajit1433@iiitd.ac.in> <ajit1433@iiitd.ac.in> <ajit1433@iiitd.ac.in>

IIIT- Delhi Office: 91-11-26907400-7404 (5 lines) / Fax: 91-11-26907405
Okhla Industrial Estate,Phase III (Near Govind Puri Metro Station) New
Delhi, India - 110020

ajit1433, 2016-08-06:
Hi,

Following are the details:

**Android OS Version:** 
ro.build.version.sdk=17
ro.build.version.codename=REL
_ro.build.version.release=4.2_
......
ro.product.name=rk2928sdk
ro.product.device=rk2928sdk
ro.product.board=rk2928sdk
ro.product.cpu.abi=armeabi-v7a
ro.product.cpu.abi2=armeabi
....
ro.board.platform=rk2928
# ro.build.product is obsolete; use ro.product.device
ro.build.product=rk2928sdk

**Kernel Version:**
_Linux version 3.0.36+ (gcc version 4.6.x-google 20120106 (prerelease) (GCC) )_

ajit1433, 2016-08-16:
Figured it out...

Initially **.wMaxPacketSize was set dynamically** in **kernel/drivers/usb/gadget/f_hid.c**

After modification:
/* High-Speed Support */
static struct usb_endpoint_descriptor hidg_hs_in_ep_desc = {
	.bLength		= USB_DT_ENDPOINT_SIZE,
	.bDescriptorType	= USB_DT_ENDPOINT,
	.bEndpointAddress	= USB_DIR_IN,
	.bmAttributes		= USB_ENDPOINT_XFER_INT,
	**.wMaxPacketSize	= 64,**
	.bInterval		= 4, /* FIXME: Add this field in the
				      * HID gadget configuration?
				      * (struct hidg_func_descriptor)
				      */
};
...
....
/* Full-Speed Support */
static struct usb_endpoint_descriptor hidg_fs_in_ep_desc = {
	.bLength		= USB_DT_ENDPOINT_SIZE,
	.bDescriptorType	= USB_DT_ENDPOINT,
	.bEndpointAddress	= USB_DIR_IN,
	.bmAttributes		= USB_ENDPOINT_XFER_INT,
	**.wMaxPacketSize	= 64,**
	.bInterval		= 10, /* FIXME: Add this field in the
				       * HID gadget configuration?
				       * (struct hidg_func_descriptor)
				       */
};

closed 2016-08-16 by ajit1433