pelya/android-keyboard-gadget #55: Galaxy S7 Edge SM-G935F Patch Problem

krakz999, 2016-09-22:
Hello

I'm having problems with patching my kernel for my device.

Log:

can't find file to patch at input line 5
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/drivers/usb/gadget/Makefile b/drivers/usb/gadget/Makefile
|index 581a5ae..6be0aa5 100644
|--- a/drivers/usb/gadget/Makefile
|+++ b/drivers/usb/gadget/Makefile
--------------------------
File to patch: 
Skip this patch? [y] y
Skipping patch.
1 out of 1 hunk ignored
can't find file to patch at input line 18
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/drivers/usb/gadget/android.c b/drivers/usb/gadget/android.c
|index 4de9562..b26ea34 100644
|--- a/drivers/usb/gadget/android.c
|+++ b/drivers/usb/gadget/android.c
--------------------------
File to patch: 
Skip this patch? [y] y
Skipping patch.
4 out of 4 hunks ignored
can't find file to patch at input line 91
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/drivers/usb/gadget/f_hid.c b/drivers/usb/gadget/f_hid.c
|index 83a266b..236bdf4 100644
|--- a/drivers/usb/gadget/f_hid.c
|+++ b/drivers/usb/gadget/f_hid.c
--------------------------
File to patch: 
Skip this patch? [y] y
Skipping patch.
13 out of 13 hunks ignored
patch: **** Can't create temporary file drivers/usb/gadget/f_hid.h.oEgkyle : No such file or directory

How should I fix it? Thanks

jcadduono, 2016-11-12:
look up guides on applying patches, but anyways, https://github.com/jcadduono/android_kernel_samsung_universal8890/commit/6ecf87370a4836475055c9913e370ef8f8d4882d.patch

however, it doesn't work, see comments at: https://github.com/jcadduono/android_kernel_samsung_universal8890/commit/6ecf87370a4836475055c9913e370ef8f8d4882d

Drolmen, 2017-11-17:
I meet this issue. Just move  file "xxx.patch" to kernel path

conrad-heimbold, 2017-12-16:
Closing because 2 solutions have been suggested and there was no answer since 1 year.

closed 2017-12-16 by conrad-heimbold