pelya/android-keyboard-gadget #67: What is the version that i should use for this kernel ?

alex9099, 2016-12-14:
Hi, it seems like i have to compile the kernel myself to use this "mod", but i cant find what is the version of the kernel for my phone, also i dont know which patch to use :( 

The kernel for my phone is this one https://github.com/bq/aquaris-E5FHD/

gmkbenjamin, 2016-12-17:
The Makefile gives you version info: 
VERSION = 3
PATCHLEVEL = 10
SUBLEVEL = 72

Try the 3.10 patches.

alex9099, 2016-12-17:
Thanks :) , just for curiosity, can the phone be hard bricked by changing / patching the kernel?

gmkbenjamin, 2016-12-17:
Not likely if it's soft bricked or in bootloop try reflashing the stock kernel or a different working kernel.

Not sure exactly which kernel will work with your phone. Try them one by one maybe? Use the "--check" flag when trying to apply the patch. If it doesn't give you any errors then it's a good indication that it might work with your phone.

alex9099, 2016-12-17:
> Use the "--check" flag 
Where?

gmkbenjamin, 2016-12-17:
Assuming you are using git apply to patch?

alex9099, 2016-12-17:
It gave me this output 
`error: patch failed: drivers/usb/gadget/Makefile:1
error: drivers/usb/gadget/Makefile: patch does not apply
error: patch failed: drivers/usb/gadget/android.c:43
error: drivers/usb/gadget/android.c: patch does not apply`

It does not seem like it wants to work :(

alex9099, 2016-12-17:
Just compiled the kernel for my phone following the instructions of my phone kernel github page with the patch applied, now, do i need to do the 

> To compile boot.img, launch commands: 

part?

alex9099, 2016-12-17:
I just tried the zImage with fastboot but it just shows "booting..." on terminal and "USB transfer OK" on phone screen :( that means that it is not compatible or i did something wrong?

conrad-heimbold, 2017-11-07:
- original question has been answered. 
- device codename: vegetafhd
- how to compile BQ Aquaris E5 4G (not very different to Aquaris E5 FHD): 
https://wiki.lineageos.org/devices/vegetalte/build (**adapt** to **vegetafhd** , which is your device)
- how to install LineageOS on BQ Aquaris E5 4G: https://wiki.lineageos.org/devices/vegetalte/install

closed 2017-11-07 by conrad-heimbold

alex9099, 2017-11-08:
i think i was doing something wrong, meanwhile i did have success on it, i think i was flashing the wrong file or just generating the image in a wrong way :/ btw the e5 4g uses a different soc from the fhd