pelya/android-keyboard-gadget #72: Some problem for x3 kernel 3.10.72

viscropst, 2017-01-28:
I using the patch for jiayus3 to my kernel,but the phone has not display something about /dev/hidg* and no kernel message about this.

conrad-heimbold, 2017-11-07:
Do you have root access? If not, you cannot see anything inside /dev.

viscropst, 2017-11-10:
yes

conrad-heimbold, 2017-12-16:
Did you get any "avc denied"-message in logcat?

conrad-heimbold, 2017-12-16:
1. Connect your smartphone via USB to your PC/Laptop. 
2. Check output of ``dmesg`` on your PC/Laptop. It should report something like: 
```
[136357.754901] usb 2-1: new high-speed USB device number 21 using xhci_hcd
[136357.895874] usb 2-1: New USB device found, idVendor=18d1, idProduct=4ee7
[136357.895878] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[136357.895880] usb 2-1: Product: Moto G Play
[136357.895882] usb 2-1: Manufacturer: Motorola
[136357.895883] usb 2-1: SerialNumber: ZY223RM5JZ
[136357.897718] input: Motorola Moto G Play as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.1/0003:18D1:4EE7.001A/input/input42
[136357.955685] hid-generic 0003:18D1:4EE7.001A: input,hidraw1: USB HID v1.01 Keyboard [Motorola Moto G Play] on usb-0000:00:14.0-1/input1
[136357.956856] input: Motorola Moto G Play as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.2/0003:18D1:4EE7.001B/input/input43
[136357.957115] hid-generic 0003:18D1:4EE7.001B: input,hidraw2: USB HID v1.01 Mouse [Motorola Moto G Play] on usb-0000:00:14.0-1/input2
```
If you don't get a dmesg-message like that, there must either be some USB connection problem; or your phine does not get recognized correctly.

viscropst, 2017-12-17:
I'll try it on Mi-4c,Thanks! My LeEco x3 is broken and I'll never use it.