pelya/android-keyboard-gadget #75: App not working with LineageOS 14.1

ShapeShifter499, 2017-02-02:
The app hangs with the Moto X Pure patch in place. It sticks at the message "Opening /dev/hidg0 and /dev/hidg1"  and does not display a keyboard. I know the patch works because another app, the "Keepass2 USB Keyboard Plugin", does work and inputs successfully.  https://play.google.com/store/apps/details?id=th.in.whs.k2ausbkbd

Whoopie, 2017-03-19:
You're probably missing the sepolicy stuff, see README.md.

ShapeShifter499, 2017-03-19:
Thank you for pointing that out, I pushed some edits to https://review.lineageos.org/#/c/164459 and https://review.lineageos.org/#/c/166246    I'm just waiting for everything to be merged.

meiser79, 2017-04-23:
It's not working even with the changes mentioned in the README.md. But as I don't see any sepolicy denials in logcat, I can't narrow down the issue. Any hints are appreciated!

ShapeShifter499, 2017-04-28:
@meiser79 I had some help from another LineageOS user by the name of "Whoopie".  He suggested some edits that I uploaded to https://review.lineageos.org/#/c/166246     Can you tell if those work?