pelya/android-keyboard-gadget #76: pls help me, kernel for  XIAOMI3TD

domald, 2017-02-08:
I want a custom kernel for xiaomi 3 pisces.

kali nethunter   HID cannot use

conrad-heimbold, 2017-10-23:
If you say "I want", **nobody** will **ever** help you. 
If you say "I am searching for" or "I would like to have", somebody **might** consider helping you.

conrad-heimbold, 2017-11-07:
- DEVICE_CODENAME: pisces (see http://en.miui.com/thread-252582-1-1.html ) 
- SOC: Nvidia Tegra 4 AP40 (see https://www.devicespecifications.com/en/model/69422983 )
- KERNEL_SOURCE: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/pisces-kk-oss =>  or https://android.googlesource.com/kernel/tegra/ 

1. Check the Makefile in the kernel src to get the kernel version of your device. 
2. Search for a suitable patch and apply it. 
3. Compile your kernel again with make.

closed 2017-11-07 by conrad-heimbold