pelya/android-keyboard-gadget #78: source code fo USB-Keyboard.apk

shellbombs, 2017-02-16:
I have a phone with a ROM which is not in your support list, when i executed the following command:

echo 0 > /sys/class/android_usb/android0/enable
echo mtp,adb,hid > /sys/class/android_usb/android0/functions
echo 1 > /sys/class/android_usb/android0/enable

then, /dev/hidg0 appeared, and i can use hid-gidgest-test to send key events, however there was no /dev/hidg1, so i can't send mouse events, and i want to use the USB-Keyboard.apk, it just showed " you will need a custom kernel *** ", i think that maybe the app check both /dev/hidg0 and /dev/hidg1, i want to patch it to use only keyboard function, so do you have the source code of USB-Keyboard.apk?

pelya, 2017-02-16:
App sources are here:
https://github.com/pelya/android-keyboard-gadget/tree/master/remote-client
You will also need SDL repository to build it:
https://github.com/pelya/commandergenius

shellbombs, 2017-02-17:
thank you very much, really appreciate it.

closed 2017-02-17 by shellbombs