pelya/android-keyboard-gadget #82: Cyanogen 13 Android 6.0.1 kernel 3.4.110

sku77, 2017-04-10:
Can you please help me with the patch? I really need to make it work badly.

Thanks in advance Pelya.

conrad-heimbold, 2017-10-23:
We can't help you if you don't give us enough information. 
What is the error message exactly? 
1. Find out the **device codename** of your LG v10. Example: For "Motorola Moto G4 Play", it is "harpia"! 
2. Find out the **SoC codename** of your LG v10. Example: For "Motorola Moto G4 Play", it is "MSM8916" by Motorola, so "motorola_msm8916"! 

Search in your android settings > about phone , lineageos wiki, lineageos download section, wikipedia.

conrad-heimbold, 2017-12-21:
A detailed step-by-step guide on how to compile lineageos (successor for cyanogenmod) can be found at
https://wiki.lineageos.org/devices/i9300/build . Replace "i9300" in this link with the codename of your device! 
You can find out the code name of your device by visiting https://wiki.lineageos.org/devices . 

Closing this issue because there has been no answer since 8 months.

closed 2017-12-21 by conrad-heimbold