pelya/android-keyboard-gadget #84: kernel-3.4.patch source code

michael86, 2017-04-20:
@pelya I'm looking at the source code for the 3.4 patch. I'm in no way a developer of any sort, can read HTML and bits of PHP that's about it. Am I right in assuming the code within the 3.4 kernel patch is a mixture of C and (maybe) ruby?? THE @@ lines are making me think ruby?? 

From what I can gather, the files job is to remove any lines in red and append any lines in green using the other lines in basic black font to regex where to place the green? 

When I tried to patch, chunk 1 and 4 were successful but 2 and 3 failed. So to finish the patch manually I'd add the 2nd and 3rd (green) lines into the relevant file? Whilst removing any in red? 

Also out of curiosity, does the @@ lines signify where the relevant code is meant to be? 

If this is the case then that's another step closer!

benoit-pierre, 2017-04-20:
@michael86: it's C, and here the patch format: https://en.wikipedia.org/wiki/Diff_utility#Unified_format

michael86, 2017-04-20:
@benoit-pierre that has got to be the quickest response I've ever received. Thank you for the wiki, going to have a read of it now.

closed 2017-04-20 by michael86

pelya, 2017-04-20:
Kernel 3.4 is really old, you can try the patch that matches your kernel
version better, like 3.10.

michael86, 2017-04-20:
@pelya Cheers for that, just had a quick bash at 3.1 and every hunk was succesful except hunk 3. Worked a lot better than the 3.4 patch at least. Going to go do some food, then I'll try some of the other patches, if any work 100% I'll update you incase you want to mention it in your repo. 

the kernel source I'm using is: https://github.com/CyanogenMod/android_kernel_htc_m7/tree/cm-13.0
along with cm13 rom for the htc m7.