pelya/android-keyboard-gadget #85: compile issue

michael86, 2017-04-21:
@pelya, new one for you ;) 

Applied the patchs. Everything showed as success except two lines which were just adding the libaries. Done that part manually.

Now when I compile, I receive the following error when it comes to:  drivers/usb/gadget/android.o

There error is as follows: 

`drivers/usb/gadget/android.c: In function 'functions_show':
drivers/usb/gadget/android.c:2093:2: warning: passing argument 2 of 'android_enable_function' from incompatible pointer type [enabled by default]
error, forbidden warning: android.c:2093
scripts/Makefile.build:307: recipe for target 'drivers/usb/gadget/android.o' failed
make[3]: *** [drivers/usb/gadget/android.o] Error 1
scripts/Makefile.build:443: recipe for target 'drivers/usb/gadget' failed
make[2]: *** [drivers/usb/gadget] Error 2
scripts/Makefile.build:443: recipe for target 'drivers/usb' failed
make[1]: *** [drivers/usb] Error 2
Makefile:952: recipe for target 'drivers' failed
make: *** [drivers] Error 2
`
The code that is around line 2093 is:

`android_enable_function(dev, "hid");`

The whole block is as follows: 

```
functions_show(struct device *pdev, struct device_attribute *attr, char *buf)
{
	struct android_dev *dev = dev_get_drvdata(pdev);
	struct android_configuration *conf;
	struct android_usb_function *f;
	char *buff = buf;

	mutex_lock(&dev->mutex);

	list_for_each_entry(conf, &dev->configs, list_item) {
		if (buff != buf)
			*(buff-1) = ':';
		list_for_each_entry(f, &conf->enabled_functions, enabled_list)
			buff += snprintf(buff, PAGE_SIZE, "%s,", f->name);
	}
	/* HID driver always enabled, it's the whole point of this kernel patch */
	android_enable_function(dev, "hid");

	mutex_unlock(&dev->mutex);

	if (buff != buf)
		*(buff-1) = '\n';
	return buff - buf;
}
```
Any help would be appreciated.

pelya, 2017-04-21:
Looks like you have added the call to android_enable_function(dev, "hid")
to the wrong function - it should be inside functions_store() not inside
functions_show().
The amount of arguments for android_enable_function() is different for
different kernel versions, try to use another patch, for example
kernel-3.10-nexus6.patch

michael86, 2017-04-21:
okay thanks, I didn't apply the android_enable_function, the patch done that part on it's own. I'll have a look at the other patches now. Will report back any issues or if it works.

michael86, 2017-04-22:
@pelya you sir are a true gentlemen! I've done it :100: 

Both hidg0 and hidg1 are showing up! If you ever come to the UK feel free to hit me up and I'll buy you a pint! 

Before I stop bugging you, if I make a text document detailing what patch I had to use and the steps to finish the patch manually, could I upload it to your repo?

pelya, 2017-04-22:
Sure, make a pull request and I'll merge it.

conrad-heimbold, 2017-12-16:
Closing because has been fixed.

closed 2017-12-16 by conrad-heimbold