pelya/android-keyboard-gadget #87: Kernel Patch for 3.10.105

dhanush-raj, 2017-05-09:
@pelya 
Hi,
I used all the kernel patches you provided to patch my kernel. But, I failed the patching itself was not sucessfull.
Like when i patched the kernel-3.10-nexus9.patch I got failed message like this below.

patching file drivers/usb/gadget/Makefile
Hunk #1 FAILED at 1.
Hunk #2 succeeded at 68 (offset -4 lines).
1 out of 2 hunks FAILED -- saving rejects to file drivers/usb/gadget/Makefile.rej
patching file drivers/usb/gadget/android.c
Hunk #1 FAILED at 43.
Hunk #2 succeeded at 2997 with fuzz 1 (offset 1591 lines).
Hunk #3 FAILED at 1457.
Hunk #4 succeeded at 3324 with fuzz 1 (offset 1672 lines).
2 out of 4 hunks FAILED -- saving rejects to file drivers/usb/gadget/android.c.rej
patching file drivers/usb/gadget/f_hid.c
Hunk #1 succeeded at 17 with fuzz 2.
Hunk #2 succeeded at 66 (offset 4 lines).
Hunk #3 succeeded at 238 (offset 23 lines).
Hunk #4 succeeded at 313 (offset 23 lines).
Hunk #5 succeeded at 332 (offset 23 lines).
Hunk #6 succeeded at 377 (offset 23 lines).
Hunk #7 succeeded at 487 (offset 75 lines).
Hunk #8 succeeded at 504 (offset 73 lines).
Hunk #9 FAILED at 638.
Hunk #10 succeeded at 733 (offset 23 lines).
Hunk #11 succeeded at 753 (offset 23 lines).
Hunk #12 FAILED at 777.
Hunk #13 FAILED at 831.
3 out of 13 hunks FAILED -- saving rejects to file drivers/usb/gadget/f_hid.c.rej
patching file drivers/usb/gadget/f_hid.h
patching file drivers/usb/gadget/f_hid_android_keyboard.c
patching file drivers/usb/gadget/f_hid_android_mouse.c

hey, please help me and give me a working kernel patch, and i will test it and report it to you. Then you can add it to your repo.
The kernel source for my kernel is https://github.com/Jleeblanch/android_kernel_motorola_msm8952
Thank you in advance.
You did a great job creating this...

conrad-heimbold, 2017-11-07:
- Do you have a Motorola Moto G4 (athene)? 
- Did you try to manually fix the patch errors? 
For every FAILED ; so for every $FILENAME.rej file; open 
$FILENAME and $FILENAME.rej ; and manually fix the patches

conrad-heimbold, 2017-12-16:
Closing, because I did not get any answer.

closed 2017-12-16 by conrad-heimbold