pelya/android-keyboard-gadget #90: Add please patch for linux kernel 3.0.101

Staskkk, 2017-06-28:
I have Galaxy S3 mini I8190 with custom ROM OS 4.4.4 and kernel 3.0.101 from here http://novafusion.pl/downloads/ . Is it possible to do patch for this kernel?

tim241, 2017-07-22:
We need the kernel source code to make a patch, link us to the source code and then people could make a patch.

conrad-heimbold, 2017-12-16:
You can try the general kernel patch for [kernel version 3.01](https://github.com/pelya/android-keyboard-gadget/blob/master/patches/existing_tested/by-generic-kernel-version/generic_kernel_3.01.patch)

I8190 uses some special SoC (ST-Ericsson NovaThor U8500). We don't have a patch specifically for that architecture, because it seems to be very rare.

conrad-heimbold, 2017-12-21:
No answer since 5 days, therefore closing.

closed 2017-12-21 by conrad-heimbold