pelya/android-keyboard-gadget #93: Android.o is not compiling

dhanush-raj, 2017-07-22:
My ROM's Kernel Developer @Jleeblanch has applied the patch according to you @pelya but he is not able to compile it successfully. What he says is our kernel is partially pacthed already..The kernel source is located at https://github.com/Jleeblanch/android_kernel_motorola_msm8952 "kali branch".
You @pelya and other's might look at it and say what the error it has please...

brianirungu, 2017-07-26:
It won't  work

brianirungu, 2017-08-01:
> @brianirungu <https://github.com/brianirungu>
> Can I know why... please
>
> *Sent from my Motorola Moto G4 Plus using FastHub
> <https://play.google.com/store/apps/details?id=com.fastaccess.github>*

I could not  logg in

dhanush-raj, 2017-08-01:
@brianirungu what you are saying say it clearly....