pelya/android-keyboard-gadget #94: App run but pc not recognize hid device.

TThuanAn, 2017-08-16:
HTC m8 3.4.0 stock kernel & 3.4.10 cm kernel have a same problem.
App detect /dev/hid0 & hid1 and run perfectly. but pc does not recognize hid device. and app can't control them.

dhanush-raj, 2017-08-23:
Which Operating system are you using..Windows or Linux

TThuanAn, 2017-08-29:
I'm test both of operating system on 4 different pcs, laptops.
2 kernels patch successful.  but pc, laptops don't recognize devices. plug phone in and adb/MTP connection show up. not detect any HID devices.

conrad-heimbold, 2017-10-23:
What is the output of dmesg on your linux pc when connecting the phone with patched kernel via usb? 
On my PC, it was: 
```
[136357.754901] usb 2-1: new high-speed USB device number 21 using xhci_hcd
[136357.895874] usb 2-1: New USB device found, idVendor=18d1, idProduct=4ee7
[136357.895878] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[136357.895880] usb 2-1: Product: Moto G Play
[136357.895882] usb 2-1: Manufacturer: Motorola
[136357.895883] usb 2-1: SerialNumber: ZY223RM5JZ
[136357.897718] input: Motorola Moto G Play as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.1/0003:18D1:4EE7.001A/input/input42
[136357.955685] hid-generic 0003:18D1:4EE7.001A: input,hidraw1: USB HID v1.01 Keyboard [Motorola Moto G Play] on usb-0000:00:14.0-1/input1
[136357.956856] input: Motorola Moto G Play as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.2/0003:18D1:4EE7.001B/input/input43
[136357.957115] hid-generic 0003:18D1:4EE7.001B: input,hidraw2: USB HID v1.01 Mouse [Motorola Moto G Play] on usb-0000:00:14.0-1/input2
```
With my patched phone kernel, it is working.

conrad-heimbold, 2017-12-19:
Do you have usb-3 or usb-2 ports at your pc/laptop? 
usb-3 will definitely **not** work, usb-1 and usb-2 should. If the usb port is blue inside, it's usb-3. Otherwise it's black.