pelya/android-keyboard-gadget #95: How To Check

doggukan, 2017-08-23:
i have installed usb gadget enabled kernel but how can i check if it works or not ? there is no hid devices and i cant edit the functions file

dhanush-raj, 2017-08-23:
Check in the kali nethunter app for hid interfaces. If dev/hidg0 and dev/hidg1 is present or not.
And one more thing your ROM Developer should add the SEPOLICY commits to it. Otherwise your SElinux mode must be set to permissive..

doggukan, 2017-08-23:
there is no hidg interfaces

dhanush-raj, 2017-08-24:
Are you sure @doggukan  that your kernel is patched according to pelya`s guide.
if then set selinux mode to permissive or in terminal app write setenforce 0. And Check once again

doggukan, 2017-08-24:
setenforce giving error

doggukan, 2017-08-24:
on kernel changelog and info .there is hid support added and selinux  is permissive i dont why it didnt workd

dhanush-raj, 2017-08-25:
Is your ROM rooted with magisk or SuperSu

_Sent from my Motorola Moto G4 Plus using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

conrad-heimbold, 2017-11-07:
what does logcat report?

conrad-heimbold, 2017-12-16:
No answer, therefore closing.

closed 2017-12-16 by conrad-heimbold