pelya/android-keyboard-gadget #97: Galaxy Tab 2 7.0 p3100

tijesef, 2017-09-12:
Can anyone compile kernel for this device Galaxy Tab 2 7.0 p3100. Any firmware, just tell me

tijesef, 2017-10-17:
Please answer somebody!

conrad-heimbold, 2017-10-23:
We can't help you if you don't give us enough information. 
1. Find out the **device codename** of your LG v10. Example: For "Motorola Moto G4 Play", it is "harpia". 
2. Find out the **SoC codename** of your LG v10. Example: For "Motorola Moto G4 Play", it is "MSM8916" by Motorola, so "motorola_msm8916". 

Search in your android settings > about phone , lineageos wiki, lineageos download section, wikipedia. 

Also, we are **not** obligated to do **anything'** for you. Please find the kernel yourself by searching for "<SoC codename> android kernel", downloading it, compiling it with "make" .

conrad-heimbold, 2017-11-06:
DEVICE_CODENAME = espresso3g
SOC = omap4
LINEAGEOS_KERNEL_SOURCE = https://github.com/LineageOS/android_kernel_ti_omap4/blob/cm-14.1/Makefile 
LINEAGEOS_KERNEL_VERSION = 3.0

Please try out the patches for kernel version 3.1 , 3.4 and 3.18 yourself.

closed 2017-11-06 by conrad-heimbold