pelya/android-keyboard-gadget #98: Add support for OnePlus 5

keeganjk, 2017-09-19:
I have a OnePlus 5 and I would like to use it as a keyboard for a computer. However, I need the Android keyboard gadget. How would I install it on the OnePlus 5?

conrad-heimbold, 2017-10-23:
Oneplus 5 has the SoC MSM8998 (see https://de.wikipedia.org/wiki/OnePlus#OnePlus_5 and https://en.wikipedia.org/wiki/List_of_Qualcomm_Snapdragon_systems-on-chip ); same as Xperia XZ Premium. 
Kernel source for msm8998 is located here: https://github.com/AndroPlus-org/android-kernel-sony-msm8998 . 
Try out the bare-bones kernel patches for kernel version 3.1 , 3.4 and 3.18 for your device and compile the kernel again afterwards! 
Steps: 
1. Check which kernel version is running on your device: Go to Android settings > About Phone > kernel version . Is it 3.1 , 3.4 or 3.18 ? 
2. If yes, try the kernel patches for 3.1 , 3.4 or 3.18 with git apply *.patch . Before, select the right kernel version with git check
3. Build the kernel again with make .

keeganjk, 2017-10-23:
My kernel is 4.4.1,I don't know anything about kernel compilation.

conrad-heimbold, 2017-11-06:
Currently, we don't have patches for 4.4 kernels. 
Please follow kernel version 4.4 progress in issue #101 .

closed 2017-11-06 by conrad-heimbold