pelya/android-keyboard-gadget #99: Patch boot.img ?

Wardenfar, 2017-09-22:
Can i patch the boot.img (MotoG4) because i have truc tout compile the kernel but i fail

conrad-heimbold, 2017-10-23:
We can't help you if you don't tell us the exact problem. What was the error message during compiling? 
Did you try to manually patch the source? 
Do you have Moto G4 (athene), Moto G4 Play (harpia), or even Moto G4 Plus? They are all different...
Kernel source for athene is located here: https://github.com/LineageOS/android_kernel_motorola_msm8952 .

conrad-heimbold, 2017-10-23:
Which kernel is running on your device? 
Go to Android settings > About Phone > kernel version to check. 
If it is 3.1 , 3.4 or 3.18 , try the patch files given in the repo.

Wardenfar, 2017-10-23:
3.1

conrad-heimbold, 2017-10-23:
1. Download the src from the link with git: git clone ...
2. Download the patch file. 
3. Go to the root of your kernel src . 
4. Type git apply <patch-file-dir> 

When errors occur, try to fix them manually.

Wardenfar, 2017-10-23:
Thanks I m going to try

conrad-heimbold, 2017-11-07:
How to build LineageOS (including the kernel) for Motorola Moto G4: 
https://wiki.lineageos.org/devices/athene/build

closed 2017-11-07 by conrad-heimbold