pelya/android-keyboard-gadget #101: support for 4.4.x branch?

aw1cks, 2017-10-09:
how can I go about patching a 4.4.x branch kernel?

goshil, 2017-10-24:
+1 Yeah, really, 4.x has different structure, some files are patchable, some - not. If somebody will guide in applying this patchset to 4.x, it will be great!

jhoola86, 2017-11-27:
Is source available for 4.x kernel versions ?

I couldn't find them in git repo

conrad-heimbold, 2017-12-17:
@jhoola86 : no, unfortunately not (yet).

aw1cks, 2017-12-27:
any update on this? is it planned at all?

conrad-heimbold, 2017-12-27:
If anybody wants to create a patch for kernel 4.4 by themself, a quote from issue #12 on how @pelya invented the code: 

## app code: 
> First, there is a kernel patch - this is dark magic, without which nothing
will work. This patch adds two new devices to your Android system,
/dev/hidg0 for keyboard, and /dev/hidg1 for mouse.
If you want to add HID joystick for example, you need to modify
hid_function_bind_config() and hid_function_init(), and another struct
ghid_device_android_joystick, with proper USB descriptor.
The best way to get USB descriptor is to buy USB analyzer, and copy data
from existing USB joystick. You can also read USB specification:
http://en.wikipedia.org/wiki/USB_human_interface_device_class

> Then you need Android application, which will write data to these devices.
It doesn't matter if it's written in C or Java, as long as it writes valid
data to these files. Keyboard/mouse protocol description is here:
https://github.com/pelya/android-keyboard-gadget#how-it-works
> My Android app is written in C++ and uses SDL library. gfx.cpp contains
low-level graphics functions, gui.cpp is a GUI widgets implementation,
flash-kernel.cpp is for flashing kernel on Nexus 7, input.cpp is for
opening device files and writing data to them, main.cpp is, well, main(),
touchpad.cpp is the touchpad implementation, and scancodes.cpp contains
keyboard scancode table.

## kernel patches: 
> I do not know all details myself, because the kernel is huge, I only know
how my code works.
> 
> First, I've read this documemt, it contains half-baked example of what I
needed:
> 
> https://github.com/torvalds/linux/blob/master/Documentation/usb/gadget_hid.txt
> 
> Then I've scanbed through Android USB gadget implementation, where
functions are added and removed dynamically:
> 
> https://android.googlesource.com/kernel/tegra/+/android-tegra3-grouper-3.1-jb-mr2/drivers/usb/gadget/f_adb.c
> 
> https://android.googlesource.com/kernel/tegra/+/android-tegra3-grouper-3.1-jb-mr2/drivers/usb/gadget/android.c
> 
> HID gadget already has it's own driver, I needed only to provide USB HID
descriptor:
> 
> https://android.googlesource.com/kernel/tegra/+/android-tegra3-grouper-3.1-jb-mr2/drivers/usb/gadget/f_hid.c
> 
> Then I fixed compilation errors, run, debugged crashes, usual stuff.

closed 2022-03-02 by aw1cks