pelya/android-keyboard-gadget #104: Steps for Samsung

Eldardvr, 2017-10-27:
Guys need help in the steps to compile the boot.img for this device. I've been trying to do this for about three days now, I suppose I do not have enough knowledge to work with the software&terminal, I am so tired: 

MANUFACTURER: Samsung
DEVICE_NAME: Galaxy-J2
DEVICE_CODENAME: J2LTE
MODEL_NUMBER: SM-J200H
SOC_NAME: Exynos 3475, Cortex-A7
Processor Model:  Mali-T720
MANUFACTURER_LINUX_KERNEL_VERSION: 3.10.49

tijesef, 2017-10-30:
You are using Samsung. Right? So use 3.18 patch. I am have 3.0.101 kernel for all roms from latest samsing stock 4.2.2 to 6.0.1. I used the 3.01 patch and everything worked!

conrad-heimbold, 2018-03-28:
Next steps are: 
1. download the source for your kernel from your link (https://github.com/j23g/android_kernel_samsung_j23g) 
2. Go to the root of the linux kernel tree (where you unpacked / downloaded the src) 
3. Put the patch file there as well (download it as well)
4. Execute ``patch -p1 < name-of-patch-file``