pelya/android-keyboard-gadget #109: mouse event not work

Drolmen, 2017-11-17:
mouse event not work at windows 10 and macOS Sierra 10.12.6 , but working well at Ubuntu. 
I use Nexus 5 with android 5.0 system. I use this [kernek for nexus 5](https://github.com/pelya/android-keyboard-gadget/blob/master/compiled/Google_Nexus-5_HAMMERHEAD_ANDROID_5.0/boot.img) .
Please help me.

Drolmen, 2017-11-24:
I have fix this issue.As follows:
adb shell
su
echo -e -n "\x0\x{X}\x{Y}\x0" > /dev/hidg1

{X} is distance at X axis, {Y} is distance at Y axis

closed 2017-11-24 by Drolmen

michaeliwaikato, 2018-04-15:
Hi there, do you think you can share how you worked this out and whether you got the mouse buttons to work please? I would very much appreciate it!

Drolmen, 2018-04-18:
@michaeliwaikato Yes, i have fixed this successful. You should directly write bytes that contain 4 byte to  dest file (/dev/hidg1). Cmd as below, not use "./hid-gadget-test"。

Drolmen, 2018-04-19:
There is a useful software named [USB Monitor](http://www.usb-monitor.com/) for widnows. You can capture the data send to windows. Hope to help you.

michaeliwaikato, 2018-05-05:
I have put some info in #135 if it is useful to anyone who finds this issue with the mouse