pelya/android-keyboard-gadget #114: My Device Doesn't Suppert HID and i dont Have hidg0 & hidg1 drivers

zenop, 2017-12-07:
Hi Gays 
The Title is shownig my problem and i will explain it more.
First of all 
My Device J2 Prime G532F
I want to add hidg0 and hidg1 to /dev
and i will try to run the payloads with juice ssh i hope it will work :(
i want to show me how to add hidg0 & hidg1 to /dev & will it work or not 
Tanks :)

Dante383, 2017-12-17:
You will have to compile the kernel for your device, as described in README.

zenop, 2017-12-21:
Ok now where i can found hidg0 and hidg1 files -?

conrad-heimbold, 2017-12-21:
If the kernel **on your android device** has USB-HID keyboard support, you can find /dev/hidg1 and /dev/hidg0 **on your android device**. 
You will also need root access and either a terminal app or adb.

Billben0012, 2020-10-18:
Hi Everyone 
I'm a new in Githib and in Linux Kernel.
Would you please tell me how to unpack/repack zIamge in my boot.img file of Android TV, so as to have access to HID files and make some modifications ??
Please I need help :D :D