pelya/android-keyboard-gadget #115: And what with 3.10 kernel?

Dante383, 2017-12-10:
I wanted to compile a kernel path for Galaxy J5 2017 (j5xnlte, MSM8916) but there is no such kernel path in the repository. What am I supposed to do?

conrad-heimbold, 2017-12-16:
Please try out 
- generic kernel version [3.04](https://github.com/pelya/android-keyboard-gadget/blob/master/patches/existing_tested/by-generic-kernel-version/generic_kernel_3.04.patch) and [3.18](https://github.com/pelya/android-keyboard-gadget/blob/master/patches/existing_tested/by-generic-kernel-version/generic_kernel_3.18.patch) patches
- [motorola_msm8916_kernel-3.10 patch](https://github.com/pelya/android-keyboard-gadget/blob/master/patches/existing_tested/by-manufacturer-and-soc/motorola__msm8916______kernel_3.10.patch )

Dante383, 2017-12-16:
@conrad-heimbold motorola_msm8916 was the closest one, but I get these while patching: 

> patching file drivers/usb/gadget/Makefile
> patching file drivers/usb/gadget/android.c
> Hunk #1 FAILED at 61.
> Hunk #2 FAILED at 2899.
> Hunk #3 FAILED at 2979.
> Hunk #4 succeeded at 3392 (offset 54 lines).
> 3 out of 4 hunks FAILED -- saving rejects to file drivers/usb/gadget/android.c.rej
> patching file drivers/usb/gadget/f_hid.c
> patching file drivers/usb/gadget/f_hid.h
> patching file drivers/usb/gadget/f_hid_android_keyboard.c
> patching file drivers/usb/gadget/f_hid_android_mouse.c

I'm using kernel from here: https://github.com/MSM8916-Samsung/android_kernel_samsung_msm8916
What am I doing wrong?

conrad-heimbold, 2017-12-16:
You are not doing anything wrong. 
Please try to patch the files manually by comparing every $FILE with its $FILE.rej

Dante383, 2017-12-16:
Okay, compiled. I'm going to flash him now, keep your fingers crossed.

conrad-heimbold, 2017-12-16:
Wait!

Dante383, 2017-12-16:
I didn't get a notification from your post, but.. guess what... yes! It works! My first compiled kernel ever, thank you stranger! <3 It feels amazing.

conrad-heimbold, 2017-12-16:
Congratulations! Could you upload the diff as a patch file? That would be really nice! 
With "diff as a patch file" I mean: 
``
git diff cm-14.1 > samsung_msm8916_kernel-3.10.patch
``

Dante383, 2017-12-16:
Sure, but are you sure that I have to compare it with cyanogenmod 14.1? Actually I have used kernel from official samsung open source page, so give me some time, I have to figure out how to make a diff out of this

conrad-heimbold, 2017-12-16:
Ok, then it is more difficult. Could you unpack the src from samsung somewhere else again and compare the two folders with diff -r folder-with-patch folder-without-patch? Thanks!

Dante383, 2017-12-16:
`diff -r kernel/drivers/gator/gator.mod.c kernel_org/drivers/gator/gator.mod.c
19,153d18
< static const struct modversion_info ____versions[]
< __used
< __attribute__((section("__versions"))) = {
< 	{ 0x9cd374bc, __VMLINUX_SYMBOL_STR(module_layout) },
< 	{ 0x61b7b126, __VMLINUX_SYMBOL_STR(simple_strtoull) },
< 	{ 0x2d3385d3, __VMLINUX_SYMBOL_STR(system_wq) },
< 	{ 0xfa012fe7, __VMLINUX_SYMBOL_STR(tracepoint_probe_register) },
< 	{ 0x262f20a8, __VMLINUX_SYMBOL_STR(local_clock) },
< 	{ 0xfd002802, __VMLINUX_SYMBOL_STR(kmalloc_caches) },
< 	{ 0xff178f6, __VMLINUX_SYMBOL_STR(__aeabi_idivmod) },
< 	{ 0xfbc74f64, __VMLINUX_SYMBOL_STR(__copy_from_user) },
< 	{ 0x22e1ae6f, __VMLINUX_SYMBOL_STR(up_read) },
< 	{ 0x4b5366cf, __VMLINUX_SYMBOL_STR(down_killable) },
< 	{ 0x528c709d, __VMLINUX_SYMBOL_STR(simple_read_from_buffer) },
< 	{ 0x36d7b11a, __VMLINUX_SYMBOL_STR(of_find_matching_node_and_match) },
< 	{ 0x5fc56a46, __VMLINUX_SYMBOL_STR(_raw_spin_unlock) },
< 	{ 0xd6ee688f, __VMLINUX_SYMBOL_STR(vmalloc) },
< 	{ 0x67c2fa54, __VMLINUX_SYMBOL_STR(__copy_to_user) },
< 	{ 0x2e5810c6, __VMLINUX_SYMBOL_STR(__aeabi_unwind_cpp_pr1) },
< 	{ 0x97255bdf, __VMLINUX_SYMBOL_STR(strlen) },
< 	{ 0x842d2613, __VMLINUX_SYMBOL_STR(perf_event_release_kernel) },
< 	{ 0x2aceabed, __VMLINUX_SYMBOL_STR(hrtimer_forward) },
< 	{ 0xc8b57c27, __VMLINUX_SYMBOL_STR(autoremove_wake_function) },
< 	{ 0xf19e9355, __VMLINUX_SYMBOL_STR(cpu_online_mask) },
< 	{ 0x34184afe, __VMLINUX_SYMBOL_STR(current_kernel_time) },
< 	{ 0x46608fa0, __VMLINUX_SYMBOL_STR(getnstimeofday) },
< 	{ 0x64021802, __VMLINUX_SYMBOL_STR(uncached_logk) },
< 	{ 0xedce55d0, __VMLINUX_SYMBOL_STR(hrtimer_cancel) },
< 	{ 0xbed60566, __VMLINUX_SYMBOL_STR(sub_preempt_count) },
< 	{ 0x20000329, __VMLINUX_SYMBOL_STR(simple_strtoul) },
< 	{ 0x856b28d0, __VMLINUX_SYMBOL_STR(generic_delete_inode) },
< 	{ 0xd3f57a2, __VMLINUX_SYMBOL_STR(_find_next_bit_le) },
< 	{ 0x614fe568, __VMLINUX_SYMBOL_STR(dput) },
< 	{ 0xf087137d, __VMLINUX_SYMBOL_STR(__dynamic_pr_debug) },
< 	{ 0x593a99b, __VMLINUX_SYMBOL_STR(init_timer_key) },
< 	{ 0x9b96369d, __VMLINUX_SYMBOL_STR(mutex_unlock) },
< 	{ 0x81d5e61b, __VMLINUX_SYMBOL_STR(mmput) },
< 	{ 0x999e8297, __VMLINUX_SYMBOL_STR(vfree) },
< 	{ 0xd8f7428a, __VMLINUX_SYMBOL_STR(mount_nodev) },
< 	{ 0x40c7247c, __VMLINUX_SYMBOL_STR(si_meminfo) },
< 	{ 0x7d819333, __VMLINUX_SYMBOL_STR(kthread_create_on_node) },
< 	{ 0x7d11c268, __VMLINUX_SYMBOL_STR(jiffies) },
< 	{ 0x7c46233a, __VMLINUX_SYMBOL_STR(cpufreq_quick_get) },
< 	{ 0xe2d5255a, __VMLINUX_SYMBOL_STR(strcmp) },
< 	{ 0x455293f6, __VMLINUX_SYMBOL_STR(down_read) },
< 	{ 0xfe7c4287, __VMLINUX_SYMBOL_STR(nr_cpu_ids) },
< 	{ 0xe707d823, __VMLINUX_SYMBOL_STR(__aeabi_uidiv) },
< 	{ 0xfa2a45e, __VMLINUX_SYMBOL_STR(__memzero) },
< 	{ 0xd5f2172f, __VMLINUX_SYMBOL_STR(del_timer_sync) },
< 	{ 0x5f754e5a, __VMLINUX_SYMBOL_STR(memset) },
< 	{ 0x872a0d8a, __VMLINUX_SYMBOL_STR(kill_litter_super) },
< 	{ 0x51d559d1, __VMLINUX_SYMBOL_STR(_raw_spin_unlock_irqrestore) },
< 	{ 0xe9bff861, __VMLINUX_SYMBOL_STR(down_trylock) },
< 	{ 0x27e1a049, __VMLINUX_SYMBOL_STR(printk) },
< 	{ 0x18bd76a4, __VMLINUX_SYMBOL_STR(_raw_spin_trylock) },
< 	{ 0x5f46ea90, __VMLINUX_SYMBOL_STR(d_rehash) },
< 	{ 0x11c9185c, __VMLINUX_SYMBOL_STR(of_iomap) },
< 	{ 0xc917e655, __VMLINUX_SYMBOL_STR(debug_smp_processor_id) },
< 	{ 0x95b9cc84, __VMLINUX_SYMBOL_STR(get_task_mm) },
< 	{ 0x9b2f852b, __VMLINUX_SYMBOL_STR(kunmap) },
< 	{ 0xaafdc258, __VMLINUX_SYMBOL_STR(strcasecmp) },
< 	{ 0x95cb929c, __VMLINUX_SYMBOL_STR(of_find_all_nodes) },
< 	{ 0x9cc4f70a, __VMLINUX_SYMBOL_STR(register_pm_notifier) },
< 	{ 0x84b183ae, __VMLINUX_SYMBOL_STR(strncmp) },
< 	{ 0x73e20c1c, __VMLINUX_SYMBOL_STR(strlcpy) },
< 	{ 0x16305289, __VMLINUX_SYMBOL_STR(warn_slowpath_null) },
< 	{ 0xc316eaf7, __VMLINUX_SYMBOL_STR(mutex_lock) },
< 	{ 0x4c6ff041, __VMLINUX_SYMBOL_STR(add_preempt_count) },
[... 238 lines cut ...]
< 	mutex_lock(&hidg->lock);
< 	hacky_device_list_remove(hidg);
< 	mutex_unlock(&hidg->lock);
< 
746,749d659
< 	/* TODO: calling this function crash kernel,
< 	   not calling this funct ion crash kernel inside f_hidg_write */
< 	/* usb_ep_dequeue(hidg->in_ep, hidg->req); */
< 	
782c692
< int hidg_bind_config(struct usb_configuration *c,
---
> int __init hidg_bind_config(struct usb_configuration *c,
836c746
< int ghid_setup(struct usb_gadget *g, int count)
---
> int __init ghid_setup(struct usb_gadget *g, int count)
diff -r kernel/drivers/usb/gadget/Makefile kernel_org/drivers/usb/gadget/Makefile
70c70
< g_android-y			:= android.o f_hid.o
---
> g_android-y			:= android.o
`

Here it is. Probably something is wrong, but it's getting 12 AM here and with my daily amount of sleep..

conrad-heimbold, 2017-12-16:
Thank you very much!

closed 2017-12-16 by conrad-heimbold