pelya/android-keyboard-gadget #121: not working on rk3288 platform of kernel 3.10.0

liushaofang, 2017-12-27:
I am working on rk3288 platform with kernel 3.10.0, after I modify the code according to generic_kernel_3.04.patch file as provided, I got kernel panic after flash the new kernel to the device. my device log is as the attachment. could you help me to analyze the problem and fix it?
[debug2.log](https://github.com/pelya/android-keyboard-gadget/files/1589411/debug2.log)

liushaofang, 2017-12-28:
[    5.198818] hid keyboard
[    5.198830] Internal error: Oops - undefined instruction: 0 [#1] PREEMPT SMP ARM

conrad-heimbold, 2017-12-28:
Sorry, I don't have much experience in kernel development and the error message seems to be very broad / general (not that much helpful: a google search doesn't give any hints what's wrong). Maybe you can increase the debug level somehow (so that more stuff gets reported with logcat)?

liushaofang, 2018-04-26:
I have fixed the issue, anybody who have come across the same problem can refer to the following link: https://github.com/liushaofang/android_usb_keyboard, the commit was verified on rk3288 platform with kernel 3.10.0.

closed 2018-04-26 by liushaofang