pelya/android-keyboard-gadget #125: Patch not working

JumpOff7667, 2018-01-12:
Hi,
I am trying to build a kernel for my phone with keyboard patch applied but patches won't work, I tried 3.18 and won't patch at all, tried 3.04 and patched some files but some files didn't patch.

Any Idea how to get this patch to work? Can I patch files manually? If yes, how?

Kernel: 3.10
Phone: HUAWEI P8 Lite
Codename: hi6210sft
OS: Android M

conrad-heimbold, 2018-03-28:
You most likely have to patch manually. that means, you have to **compare** 
- the original file, 
- the not-completely patched file
- the diff / patch file 
... and deduce what to do.

conrad-heimbold, 2018-04-12:
Did I answer your questions?

conrad-heimbold, 2018-04-13:
You could upload your patch as a gist file to github (https://gist.github.com) and send me the link to that gist.