pelya/android-keyboard-gadget #126: Nexus 6 (XT1100) patch?

MaliciouZzHD, 2018-01-16:
I am trying to compile a custom kernel ([from here](https://github.com/LineageOS/android_kernel_moto_shamu)) for use with Kali Nethunter.

I managed to compile everything yesterday, with the [mac80211.compat08082009.wl_frag+ack_v1.patch](http://patches.aircrack-ng.org/mac80211.compat08082009.wl_frag+ack_v1.patch) patched into the kernel.

But I forgot the  android-keyboard-gadget patch :p.

I couldn't find the patch anywhere in this Git repository, so I was wondering what patch will work with a Nexus 6 (XT1100) with Kernel Version 3.10.40.

conrad-heimbold, 2018-01-23:
> So I'm guessing it hasn't been made yet? 

Yes, exactly. 
You can try any of the provided kernel_3.10 patches, I guess, which are: 

> /patches/existing_tested/by-manufacturer-and-soc/motorola__msm8974______kernel_3.10.patch
> /patches/existing_tested/by-manufacturer-and-soc/motorola__msm8916______kernel_3.10.patch
> /patches/existing_tested/by-manufacturer-and-soc/samsung___apq8084______kernel_3.10.patch
> /patches/existing_tested/by-manufacturer-and-device-codename/htc_______flounder_____kernel_3.10.patch
> /patches/existing_tested/by-manufacturer-and-device-codename/jiayu_____s3h560______kernel_3.10.patch

conrad-heimbold, 2018-04-12:
Could I help you?