pelya/android-keyboard-gadget #130: I'm getting an error when I'm trying to apply to patch:

alexandrumarcel, 2018-03-09:
Please first read CONTRIBUTING_AND_ISSUE_TEMPLATE.md and do some research about your device! 

I will close your issue without answering if you don't provide me enough information about...
1. Xiaomi Mi A1_tissot, kernel 3.18.71, Lineage OS 15.1
2. I'm getting this error `patch: **** strip count ../kernel_3.18.patch is not a number`
When I'm trying to patch using `patch -p ../kernel_3.18.patch`
3. what exactly you would like to have / do. Give kernel support for USB Keyboard

conrad-heimbold, 2018-03-19:
You have to write a number after the "p", not the file name. The number specifies how "flexible" the file should be patched.

conrad-heimbold, 2018-04-12:
You have to write ```patch -p3 ../kernel_3.18.patch``` 
Can this issue get closed?

closed 2018-04-12 by conrad-heimbold