pelya/android-keyboard-gadget #131: HUAWEI P9 LITE

sz4l3k, 2018-03-11:
Hello,

I wanted to know if it is possible to compile the kernel to be able to install it onto a Huawei P9 Lite with stock android 6.0? If it is, would you be so kind and give me some advice how to start going about it?

Thank you.

conrad-heimbold, 2018-03-19:
do you have the source for huawei p9 lite? Which SoC does your huawei p9 lite have? If you have the src, what linux version? I can't answer your question if i don't know these information.

sriram407, 2018-05-16:
I use https://github.com/codeofhonor-dev/android_kernel_huawei_hi6250/tree/o-emui5

And it has SoC Kirin 655

sriram407, 2018-05-24:
@sz4l3k hey i have finally patched the drivers.If you want then post a comment

goshil, 2018-05-24:
@sriram407 Hi! The kernel source by your link is 4.x version, have you successfully patched 4.x ? Or is this talk about 3.10.x version?

sriram407, 2018-05-25:
@goshil Hey i have patched for emui 4.1 (3.10.x) 
it is hard to patch for 4.1.x becausrye in gadget folder it doesn't have android.c file 
But also i am trying to find solution

goshil, 2018-05-25:
Thanks, got it.