pelya/android-keyboard-gadget #132: What patch should i use

fatkhur1960, 2018-03-25:
I have samsung galaxy a3(16) "a3xelte" with kernel version 3.10.xx .. can you help me to build the patch for my device? or you can suggest me what patch should i use from you current patchs? thx

conrad-heimbold, 2018-03-28:
Did you try this one: 
https://github.com/pelya/android-keyboard-gadget/blob/master/patches/existing_tested/by-manufacturer-and-soc/samsung___apq8084______kernel_3.10.patch ?

conrad-heimbold, 2018-03-28:
what model name does your galaxy a3 have? SM-A310F , SM-A310F/DS, SM-A310Y, SM-A310M or SM-A3100? Depending on that, you have a Samsung **Exynos 7578** , Qualcomm **Snapdragon 410**  or Samsung **Exynos 7870**  processor. Different processors require different linux versions and different kernel patches. 
Please checkout the wikipedia website: https://en.m.wikipedia.org/wiki/Samsung_Galaxy_A3_%282016%29

conrad-heimbold, 2018-03-28:
You can find out the model name of your galaxy a3 by going to settings > about phone.

fatkhur1960, 2018-03-30:
My model name is SM-A310F... i had try this patch https://github.com/pelya/android-keyboard-gadget/blob/master/patches/existing_tested/by-manufacturer-and-soc/samsung___apq8084______kernel_3.10.patch . But i got an error while building proccess

conrad-heimbold, 2018-03-30:
Patching almost never automatically succeeds. You almost always have to patch manually, by 
1. **comparing** every $FILE with its $FILE.rej ; and 
2. **fixing** every error.

bhasiththamis, 2019-04-22:
@conrad-heimbold  im having galaxy grand prime model:g530h codename: fortunave3g
Which patch should i use ? It has Snapdragon 410 processor ...pls fimd me the patches for my phone 
Thanks in advance:)

closed 2019-05-15 by fatkhur1960